sparkfun/SparkFun_RTK_Firmware #7: Option to create UBX file

nseidle, 2021-03-17:
After some fiddling, I determined the ubx file format is a few things:

* NMEA sentences output by a receiver
* Any RAWX messages that are enabled
* Any UBX binary messages such as receiver configuration/settings

A UBX file captured from u-center will contain lots of extra UBX messages most of which are the module's settings.

You can however create your own UBX file using the RTK Surveyor. Log NMEA and RAWX then change the TXT file extension to UBX. u-center will happily open the file and give you the ability to scroll through the timeline.

Similarly, these pseudo UBX files are good enough for RTKCONV to create OBS data (good for PPP). Knowing this, RTK Surveyor is effectively producing ubx files. Should we create an option that generates an extra file just for RAWX capture?

Option 1)

* SFE_Surveyor_210317_182802.txt created and captures all NMEA sentences that are enabled
* SFE_Surveyor_210317_182802.ubx created and captures all RAWX messages

The ubx would be smallest. It could be opened in u-center but none of the graphical stuff (sats in view, PDOP, etc) would work during timeline scrolling.

Option 2)

* SFE_Surveyor_210317_182802.txt created and captures all NMEA sentences that are enabled
* SFE_Surveyor_210317_182802.ubx created and captures all RAWX messages and NMEA.

This would allow a UBX file to be opened in u-center and show all the graphical stuff (sats in view, PDOP, etc) during timeline scrolling.

Option 3)

Give the user the ability to set the file extension - either TXT or UBX.

I really dislike the idea of creating redundant data files so I'm leaning towards option 3.

flyingburro, 2021-03-18:
Hi, Nate.

Thanks for the clarification on the file format -- this helps A LOT!  I did not realize that the raw data were encoded within that text file.  This now works really well.  The easiest thing is simply to change the suffix from ".txt" to ".ubx" (i.e., Option 3), then read it with RTKCONV.  This produces a usable RINEX file for static processing.

I don't know if a separate RAWX file is absolutely necessary.  Off the top of my head, the only reason one might create a separate RAWX file as output would be simply to reduce the file size.  I'm not sure how large a 24-hour file (say 15 second sample rate) would be.  This lower frequency sample rate and longer time frame is an example we might use if deploying these for long-term monitoring of, say, a landslide.  

I'll get back to you after I try out some longer duration deployments.  If it looks like there is a lot of wasted storage space, then, perhaps, an option to write out only the RAWX might be worth considering.  

Thanks!

Paco

nseidle, 2021-03-18:
Hi Paco! Excellent. Yea, I too thought the smaller RAWX file may be nice. Please share your experience with 15Hz at 24Hours. I suspect it will be a few GB! 

Btw - I think we may be able to record directly to OBS and skip RTKCONV. If we do, the file sizes will be pretty huge, but it will be fun to see.

I'll leave this open until I implement option 3.

flyingburro, 2021-03-18:
Hi, Nate.  I realize this may be a slight deviation from this topic, but how do I set the sample rate (at least for raw messages) to a slower rate than 1 per second (e.g., how do I set it to once every 15 seconds)?  The ESP32 configuration only accepts whole numbers (samples per second), so I can't set that slower than 1 hz.  I suspect I need to do this through u-center, correct?

nseidle, 2021-03-18:
Yes, u-center is the best place to start. You can set a rate per messages. 

[image]

Here I've set the effective rate to 1 every 4 seconds because my fix rate is 4Hz (16/4 = 4). 

I'm not sure what you're doing or why a user would need to do this but if you need a feature, please feel free to request it.

flyingburro, 2021-03-18:
Thanks, Nate.  That makes more sense -- I had been trying UBX-CFG-RATE, but that is overwritten by the settings.

What I am trying to do is slow the raw sample rate down to something less than 1 hz.  For a lot of static surveying, there is no benefit from high-frequency sample rates ... 15 second, or even 30 second, sampling is fine, but we require longer observation periods (many hours to days).  Thus, a helpful feature could include:
(1) setting slower sample rates (at the least, hard-wired options of 5, 15, and 30 seconds, but user specified sample period is more flexible)
(2) writing only RAWX output to a file (optional, but will help reduce file size)
(3) automatically cutting those data files at 24:00:00 UTC (optional, again, but keeps files small -- RINEX can always be cut/merged using TEQC, RTKLIB, etc.)

This will keep individual data files relatively small, but allow long-term, static data collection (running this thing off a larger battery or solar panel).  As I may have mentioned, one possible scenario would be deploying many of these on a landslide for monitoring.

Is this the appropriate forum for feature requests?  Or would you like me to put this in a more appropriate location?  Please let me know.  I appreciate your patience (I'm relatively new to Github).  Thanks.

nseidle, 2021-03-18:
Got it. Yes, this is the best forum. Open and issue, let me know what you want and why. 

1) yep, good feature
2) yea, programmatically hard to implement currently but can be done. Alternatively, could you just turn off the NMEA sentences? You'd have a pure RAWX file.
3) There's already a stop logging after X minutes feature. Do you need a time cutoff?

Stefal, 2021-03-22:
> 
> The ubx would be smallest. It could be opened in u-center but none of the graphical stuff (sats in view, PDOP, etc) would work during timeline scrolling.

You can keep the graphical stuff if you enable UBX-NAV-SAT, UBX-NAV-SIG. Maybe you need UBX-NAV-PVT and UBX-NAV-DOP too.

nseidle, 2021-03-22:
I stand corrected! Thanks @Stefal.

Stefal, 2021-03-23:
My opinion is that completely disabling nmea inside log file could be useful. I need it only on the bluetooth output.

nseidle, 2021-03-29:
v1.2 release candidate now creates two files: [date-time].txt for only nmea and [date-time].ubx for all/only UBX messages. Currently only RAWX and SFXBR are supported but adding more is straightforward. 

NMEA is sent over Bluetooth SPP. No UBX messages are sent over SPP, although it's possible but I'm going to wait for a request for it.

I'm going to try to implement Paco's multiple seconds between readings before I push binaries.

nseidle, 2021-03-29:
We'll have to modify the u-blox library to get <1Hz measurement rates.

nseidle, 2021-03-29:
Ok, binaries are [here](https://github.com/sparkfun/SparkFun_RTK_Surveyor/tree/release_candidate/Binaries):

* [v1.2 RC](https://github.com/sparkfun/SparkFun_RTK_Surveyor/blob/release_candidate/Binaries/For_SD_Loading/RTK_Surveyor_Firmware_v12rc.bin) for loading via SD
* [v1.2 RC combined](https://github.com/sparkfun/SparkFun_RTK_Surveyor/blob/release_candidate/Binaries/RTK_Surveyor_Firmware_v12rc_combined.bin) for loading via ESPTool

Please test when possible and let me know how it works for you. I tested the output up to 10Hz with SPP connected, NMEA logging, and RAWX/SFXBR logging. This is pretty incredible for the ESP32 to pull off. At 15Hz things start to break but it's not clear to me why. I suspect the ZED is having problems serving up all the interfaces include the USB connection I have open to watch debug output from the ZED. This is the long way of saying faster than 10Hz is possible but you'll need to not be connected over BT and/or disable NMEA logging or trim the enabled sentence count.

Stefal, 2021-04-02:
Perhaps I did something wrong, but this release candidate doesn't seem to log. I've made a reset, a new test, and my ubx file is empty.
My settings:
```
sizeOfSettings=352
rtkIdentifier=18
gnssMeasurementFrequency=8
printDebugMessages=0
enableSD=1
enableDisplay=1
frequentFileAccessTimestamps=0
maxLogTime_minutes=600
observationSeconds=60
observationPositionAccuracy=5.00
fixedBase=0
fixedBaseCoordinateType=0
fixedEcefX=0.00
fixedEcefY=0.00
fixedEcefZ=0.00
fixedLat=0.00
fixedLong=0.00
fixedAltitude=0.00
dataPortBaud=115200
radioPortBaud=57600
outputSentenceGGA=1
outputSentenceGSA=1
outputSentenceGSV=1
outputSentenceRMC=1
outputSentenceGST=1
enableSBAS=0
enableNtripServer=0
casterHost=rtk2go.com
casterPort=2101
mountPoint=bldr_dwntwn2
mountPointPW=WR5wRo4H
wifiSSID=TRex
wifiPW=parachutes
logNMEA=0
logUBX=1
logRAWX=1
logSFRBX=1
```

nseidle, 2021-04-06:
Not your fault. It turns out the SdFat library is failing when the main loop is writing to the ubx file and the serial task attempts to write to the nmea file at the same time. A quick test showed that the ESP32 built in library worked as expected. Simple change right? Then 5 hours of hacking followed.

In April-6-RC:
* SdFat replaced by SD
* System time set by GNSS
* Increase stack size for serial task to 2k

There are improvements to be made but it's stable (for me) now so I figured I'd get binaries out for your testing. The current bottleneck is SD writing for rates above 4Hz.

If you haven't tried changing the firmware using the serial menu, I'd recommend it. Load both v11 and v12rc onto your SD card, open up the serial menu, and you should be able to switch back and forth.

nseidle, 2021-04-06:
I was wrong, the ESP32 SD lib is quite comparable and is often faster than SdFat write speeds:

**Kingston 16GB**

ESP32 Lib
default = 1048576 bytes written for 6019 ms
20MHZ = 1048576 bytes written for 5584 ms
40MHz = 1048576 bytes written for 4390 ms
80MHz = 1048576 bytes written for 4400 ms = 238KB/sec

SdFat
50MHz fail
16MHz write: 195KB/Sec 
40MHz fail

**SparkX 1GB**

ESP32 Lib
80MHz = 1048576 bytes written for 5609 ms = 186kb/sec

SdFat
32MHz fail
16MHz write: 157KB/Sec

Code bits I need:
    
    SPIClass spi = SPIClass(VSPI);
    if(!SD.begin(PIN_MICROSD_CHIP_SELECT, spi, 80000000)){

PaulZC, 2021-04-07:
Just for my understanding, which version of SdFfat are you comparing against? 1.1.4 or 2.0.6?
Cheers!
Paul

nseidle, 2021-04-07:
2.0.6. And in retrospect, I think I could avoided the whole shuffle by using a semaphore for the SPI hardware. I suspect the SD lib for ESP32 has built in thread friendliness whereas SdFat may be less so.

Stefal, 2021-04-09:
I'm testing the latest release candidate.

Yesterday, the RTK Surveyor didn't log anything, I don't know why. I just got 2 very small files for nmea and ubx.
Today I have a nmea file with some holes in the recording, and the ubx file is very small with only some data without timestamp:
(Packet console viewer in U-Center)
```
...
??:??:??  L -> UBX RXM-SFRBX,  Size  56,  'Subframe Data NG'
??:??:??  L -> UBX RXM-SFRBX,  Size  56,  'Subframe Data NG'
??:??:??  L -> UBX RXM-RAWX,  Size 2040,  'Multi-GNSS Raw Measurement Data'
??:??:??  L -> UBX RXM-RAWX,  Size 2040,  'Multi-GNSS Raw Measurement Data'
??:??:??  L -> UBX RXM-RAWX,  Size 2040,  'Multi-GNSS Raw Measurement Data'
??:??:??  L -> UBX RXM-SFRBX,  Size  52,  'Subframe Data NG'
??:??:??  L -> UBX RXM-SFRBX,  Size  52,  'Subframe Data NG'
??:??:??  L -> UBX RXM-SFRBX,  Size  52,  'Subframe Data NG'
...
```

nseidle, 2021-04-10:
Thanks for testing! I'm dialing in issues but I'll need you to connect over USB (for now) to see debug messages.

Dumb question but: Did you enable logging and the NMEA and UBX sentences?

Enable sentences menu:
[image]

Enable logging menu:
[image]

What do you see when you power on? For example, with my current branch, I've added some debug messages to see file sizes:

[image]

Your RC is slightly behind but you should see the 
    
    UBX byte total/buffer: 190464 / 752

message every second or so.

nseidle, 2021-04-11:
After some thinking, I plan to blink the power level LED when log file sizes are increasing. This should, at a glance, indicate to the user that data is being logged locally.

Stefal, 2021-04-11:
> Dumb question but: Did you enable logging and the NMEA and UBX sentences?

Sure !
```
sizeOfSettings=352
rtkIdentifier=18
gnssMeasurementFrequency=4
printDebugMessages=0
enableSD=1
enableDisplay=1
maxLogTime_minutes=600
observationSeconds=60
observationPositionAccuracy=5.00
fixedBase=0
fixedBaseCoordinateType=0
fixedEcefX=0.00
fixedEcefY=0.00
fixedEcefZ=0.00
fixedLat=0.00
fixedLong=0.00
fixedAltitude=0.00
dataPortBaud=115200
radioPortBaud=57600
outputSentenceGGA=1
outputSentenceGSA=1
outputSentenceGSV=1
outputSentenceRMC=1
outputSentenceGST=1
enableSBAS=0
enableNtripServer=0
casterHost=rtk2go.com
casterPort=2101
mountPoint=bldr_dwntwn2
mountPointPW=WR5wRo4H
wifiSSID=TRex
wifiPW=parachutes
logNMEA=1
logUBX=1
logRAWX=1
logSFRBX=1
```

I've reset the F9P with U-center, and reset the RTK Surveyor (I don't know if resetting the RTK Surveyor reset the F9P too)
Then I've enable RAWX, SFRBX, nmea logging, ubx logging.

Inside the terminal, I see this:
```
UBX byte total/buffer: 1057280 / 15200
UBX byte total/buffer: 1058816 / 16368
UBX byte total/buffer: 1060864 / 14632
UBX byte total/buffer: 1062912 / 15144
UBX byte total/buffer: 1064960 / 16380
....
```
When I stop the RTK Surveyor, the micro SD contains a nmea file and a ubx file.

SFE_Surveyor_210411_124533.txt -> 880 KB
SFE_Surveyor_210411_124533.ubx -> 1070 KB

The nmea file seems ok (I didn't check I some point are missing like my last test). 
The ubx doesn't contain valid data when I open it inside U-center or If I try to convert it to gpx or pos file.

ubx file -> [SFE_Surveyor_210411_124533.zip](https://github.com/sparkfun/SparkFun_RTK_Surveyor/files/6291811/SFE_Surveyor_210411_124533.zip)

I will have to send the RTK Surveyor during the next week so this is my last testing days as there are no Surveyor in stock anymore.

Stefal, 2021-04-11:
I made another test with the 1.1 release, and when I open the nmea/ubx file with U-center, it's ok, the rawx messages have a valid timestamp.

Stefal, 2021-04-11:
> 
> 
> After some thinking, I plan to blink the power level LED when log file sizes are increasing. This should, at a glance, indicate to the user that data is being logged locally.

Isn't blinking a led only when there is a problem a better solution?

nseidle, 2021-04-12:
Using the 'r'eset option from the menu will reset both the ZED-F9P settings and Surveyor settings to default. It is fine to do, it just requires a cold start (~30s) for GNSS fix at the next power up.

We should have ~50units of the RTK Surveyor available in stock by the end of the week. We have had many challenges building enough to meet demand!

I'm able to open your ubx file:

[image]

I'm also able to convert your ubx to obs (RINEX) using RTKLib:

[image]
[image]

I took a large survey over the weekend (1GB of UBX) and successfully converted to OBS. I've submitted it to [CSRS](https://webapp.geod.nrcan.gc.ca/geod/tools-outils/ppp.php?locale=en) and expect to get the results soon. Everything is working as expected for me.

Are you expecting to see location and sat data within u-center based on the ubx file? We will (thanks to your note above!) need to enable additional UBX messages.

> You can keep the graphical stuff if you enable UBX-NAV-SAT, UBX-NAV-SIG. Maybe you need UBX-NAV-PVT and UBX-NAV-DOP too.

I'll be working on a few other items today:

* Time limiting the UBX logging
* <1Hz nav rates
* Blink BASE LED to indicate logging

Stefal, 2021-04-12:
Ok, so all my tests are wrong. I'm sorry!

nseidle, 2021-04-12:
> Ok, so all my tests are wrong. I'm sorry!

No no! Please keep reporting problems until you get what you need. What are you expecting to see within u-center when you load a ubx file from Surveyor?

My CSRS results came back oddly. I'll run some more tests tonight.

Stefal, 2021-04-12:
Don't worry, I'll continue my tests when there will be a release with more message available, and ... when I will receive another Surveyor.
Beside rawx and sfrbx, I usually use:
- nav-pvt
- nav-hpposllh
- nav-hpposecef (3D accuracy seems available only in this message)
- nav-sat
- nav-sig

BTW: I know some users would like to get the ubx messages on bluetooth because with these, it's possible to use rtkgps+. There is an alpha release which works with the F9P: https://github.com/jancelin/RtkGps

nseidle, 2021-04-12:
Ok - I've start a [separate issue](https://github.com/sparkfun/SparkFun_RTK_Surveyor/issues/14) for UBX over BT SPP. It was originally supported and is trivial to bring back.

nseidle, 2021-04-21:
Of course with some tweaking, I now see that SdFat blows the regular SD library out of the water (10x faster):

[image]

Important bits:

    SPIClass SPI1(VSPI);
    #define SD_CONFIG SdSpiConfig(PIN_MICROSD_CHIP_SELECT, DEDICATED_SPI, SD_SCK_MHZ(36), &SPI1)
    if (sd.begin(SD_CONFIG) == false) {}

36MHz is about as fast as I can push it but the write speeds are pretty amazing. I'll see if I can make it work nicely with the two open files NMEA/UBX.

nseidle, 2021-04-22:
[RC-April22](https://github.com/sparkfun/SparkFun_RTK_Surveyor/blob/release_candidate/Binaries/For_SD_Loading/RTK_Surveyor_Firmware_v12rc-April22.bin) is available for SD loading. If you need a separate zip, let me know.

tldr; Semaphores for the win. 

After a few hours of logging yesterday, I'm much happier. This needs more testing but I think I've got the instability resolved. I don't believe either library was thread safe for dual file logging. Adding file system semaphores has resolved a few issues:

* Long term logging is improved. Previously, the device would indicate logging but upon file size inspection, the files were not increasing. This was indicative of the SD class getting corrupt.
* Initial log creation was hit or miss. This was because when we attempted to create a file, the F9PSerialReadTask() would kick on (any time serial was received) again, causing issues with either SD library because we were in the midst of creating a new file.

Please kick the tires when possible.

 Known issues:

* Logging faster than 4Hz is a problem. I think there's an I2C bottleneck or something else, but this is not critical to me.
* Logging NMEA is still locked to UART1 (issue #14). I have a solution to this (thanks to @paulzc) but I haven't had the chance to implement it yet.

nseidle, 2021-05-07:
@stefal and @flyingburro v1.2 is now ready. 

* When logging successfully, Base LED will blink.
* I fixed all sorts of problems and instabilities (it's now rock solid) but I was not able to separate UBX / NMEA to two files so logging happens to a single UBX file.
* You can now choose which messages to broadcast over SPP vs log
* 10Hz logging is working great. 
* @flyingburro - You can now set fixes <1Hz

closed 2021-05-07 by nseidle