sparkfun/SparkFun_RTK_Firmware #8: Document supported ZED-F9P firmware versions & upgrade procedures?

claudobahn, 2021-03-21:
The two Sparkfun RTK surveyors I have shipped with `ZED-F9P FW 1.00 HPG 1.12` and looks like a `ZED-F9P FW 1.00 HPG 1.13` firmware update is available.

1.13 description: https://www.u-blox.com/sites/default/files/ZED-F9P-FW100-HPG113_RN_%28UBX-20019211%29.pdf (older firmware release notes are cross-referenced with links on the first page of this PDF, e.g. for 1.12)

It appears that 1.13 has made useful improvements (see section 4 of the PDF).

I'm wondering if you'd be willing to document for the benefit of others:
* how to determine a surveyor's current ZED-F9P firmware version, which basically comes down to:
> * In u-center with an established connection to the ZED-F9P, click on View -> Message View
> * Navigate to UBX -> MON -> VER
* would be nice to include firmware update instructions, as well, perhaps use this as a start: https://drotek.gitbook.io/rtk-f9p-positioning-solutions/tutorials/updating-zed-f9p-firmware
* give guidance on supported / tested firmware revisions in the repo's README? Would be nice if that were easy to find.

claudobahn, 2021-03-23:
@nseidle 

Ok, so looks like 1.13 is probably supported:
https://github.com/sparkfun/SparkFun_RTK_Surveyor/blob/09431d4dbb3c032af688cd82c24a278b4ca25dd6/Firmware/RTK_Surveyor/RTK_Surveyor.ino#L130

Although the code that checks is currently commented out: https://github.com/sparkfun/SparkFun_RTK_Surveyor/blob/09431d4dbb3c032af688cd82c24a278b4ca25dd6/Firmware/RTK_Surveyor/begin.ino#L123

Any particular reason for that?

nseidle, 2021-03-26:
Good idea! I will create a tutorial to document how to update firmware.

We did all of our initial field testing with v1.12. I too thought the upgrade would improve things. Namely, the ability to restart a survey-in was drastically needed. However, v1.13 has a bug that causes the RTK indicator to fail if SBAS is used. For our product, this LED is crucial. So it was sort of a tossup: if we push to v1.13, we have to disable SBAS, thus leading to possibly worse performance. That lead to commenting out of the warning.

The ZED-F9P performs pretty spectacularly with either v1.12 or v.13. For now, we'll leave it to the user but I will certainly write that tutorial explaining how.

nseidle, 2021-03-26:
Tutorial is live [here](https://learn.sparkfun.com/tutorials/how-to-upgrade-firmware-of-a-u-blox-gnss-receiver). Please have a look and let me know if it's sufficient.

I've added a handful of notes to the firmware page [readme](https://github.com/sparkfun/SparkFun_RTK_Surveyor/tree/master/Firmware).

closed 2021-03-26 by nseidle