sparkfun/SparkFun_RTK_Firmware #9: Accurate timestamps

adamgarbo, 2021-03-22:
Hi @nseidle,

Poking around in the RTK Surveyor code I noticed that you're using the `getTimeValid` and `getDateValid` functions for timestamping log files. A quick heads up that these functions can produce inaccurate timestamps (see: https://github.com/sparkfun/SparkFun_u-blox_GNSS_Arduino_Library/issues/4)

https://github.com/sparkfun/SparkFun_RTK_Surveyor/blob/96e89e625579a63db7ba30d3f506f862aeb7fcf1/Firmware/RTK_Surveyor/menuLog.ino#L98-L105

With the u-blox ZED-F9P you'll want to use the UBX-NAV-PVT `confirmedDate` and `confirmedTime` registers available in Protocol Versions >19.00 (e.g. M9/F9). 

I now use a callback to process NAV_PVT messages to synchronize the Artemis' RTC with the ZED-F9P and it has proven to be very reliable:

```
void processNavPvt(UBX_NAV_PVT_data_t ubx)
{
  bool dateValidFlag = ubx.flags2.bits.confirmedDate;
  bool timeValidFlag = ubx.flags2.bits.confirmedTime;
  byte fixType = ubx.fixType;

  // Check if date and time are valid and sync RTC with GNSS
  if (fixType == 3 && dateValidFlag && timeValidFlag)
  {
    // Set RTC date and time
    rtc.setTime(ubx.hour, ubx.min, ubx.sec, ubx.iTOW % 1000,
                ubx.day, ubx.month, ubx.year - 2000);

    rtcSyncFlag = true; // Set flag
    Serial.print("Info: RTC time synced to "); printDateTime();
  }
}
```

Cheers,
Adam

nseidle, 2021-03-29:
Thanks! I just began migrating RTK over to v2 of the u-blox library. I don't foresee using callbacks since I don't need to constantly know if the date/time is confirmed, only once. I'll see if I can implement it otherwise.

adamgarbo, 2021-03-29:
No worries!

In my code it's the only callback I use, so it made it really easy to synchronize the time from a separate function when needed. I typically sync the RTC when powering up for the first time, and then again every day or so, since the Artemis tends to drift a little after a while.

Keen to see how you approach it! 

```
// Acquire valid GNSS fix and sync RTC
void syncRtc()
{
  // Clear flag
  rtcSyncFlag = false;

  Serial.println("Info: Acquiring GNSS fix...");

  // Attempt to acquire a valid GNSS position fix
  while (!rtcSyncFlag && millis() - loopStartTime < gnssTimeout)
  {
    petDog(); // Reset watchdog timer
    gnss.checkUblox(); // Check for arrival of new data and process it
    gnss.checkCallbacks(); // Check if callbacks are waiting to be processed
  }
  if (!rtcSyncFlag)
  {
    Serial.println("Warning: Unable to sync RTC!");
  }

}
```

adamgarbo, 2021-04-01:
Howdy,

I also decided to ditch the callbacks! An easier way was to use `.setAutoPVT(true);` and the new `.getUnixEpoch()` function to set the time on the Artemis. I don't know if the ESP32 WROOM can set its RTC with epoch time, but this would just require calls to query the individual time elements. 

Cheers,
Adam

```
void syncRtc()
{
  unsigned long loopStartTime = millis(); // Start loop timer
  bool rtcSyncFlag = false; // Clear flag

  // Attempt to acquire a valid GNSS signal
  while (!rtcSyncFlag && millis() - loopStartTime < gnssTimeout)
  {
    // Check for UBX-NAV-PVT messages
    if (gnss.getPVT())
    {
      bool dateValidFlag = gnss.getConfirmedDate();
      bool timeValidFlag = gnss.getConfirmedTime();

      // Check if date and time are valid and sync RTC with GNSS
      if (dateValidFlag && timeValidFlag)
      {
        rtc.setEpoch(gnss.getUnixEpoch()); // Set RTC date and time
        rtcSyncFlag = true; // Set flag
      }
    }
  }
  if (!rtcSyncFlag)
  {
    Serial.println("Warning: Unable to sync RTC!");
  }
}

```

nseidle, 2021-04-06:
Thanks Adam! Implemented in April-6-RC.

closed 2021-04-06 by nseidle