sparkfun/SparkFun_RTK_Firmware #10: Logging on SD doesn't work when bluetooth set to rtcm

Stefal, 2021-03-23:
Hi,

I have a RTK Surveyor for a few days/weeks, and I trying to use it as my actual logger:
F9P uart1: ubx only, data sent to the micro SD card. Rate as fast as possible
F9P uart2: rtcm in, nmea out. Nmea output is set slower than ubx

With the rtk surveyor, it seems that the default configuration sets bluetooth to get the F9P uart1 output. When I set the internal switch from nmea to rtcm, Bluetooth seems connected to uart2... good.

In this mode, logging to the sd card doesn't work.

nseidle, 2021-03-26:
The firmware is currently setup so that anything received over the ESP32s second UART (UART1) is logged to SD (if logging is enabled). Moving the switch from NMEA to RTCM will pipe the ZED's UART2 data to the ESP32 for logging.

[image]

It's not clear to me if this is what you intend or not.

A few questions for you:

* How are you configuring the ZED-F9P? Over the RTK Surveyor menu system or u-center? The ESP32 will re-configure the ZED-F9P at every power-on-reset with the user settings that have been recorded. If you are configuring the ZED using u-center you're going to see a lot of overwritten config.
* What rate(s) are you trying to record? There's a limit to how fast the ESP32 can record to SD. I haven't dug into it yet but 4Hz is currently the tested limit.

Stefal, 2021-04-01:
Hi!

> It's not clear to me if this is what you intend or not.
I like to have 2 separate streams. One "relax" stream on bluetooth (uart2), to send the rtcm to the F9P and receive some nmea strings on my smartphone. And the second stream, with rawx/sfrbx/pvt/hpposllh/sig/... at the fastest rate possible.

>     * How are you configuring the ZED-F9P? Over the RTK Surveyor menu system or u-center? The ESP32 will re-configure the ZED-F9P at every power-on-reset with the user settings that have been recorded. If you are configuring the ZED using u-center you're going to see a lot of overwritten config.

Yes I'm using u-center, and I noticed that the RTK Surveyor overwrite some settings.

> 
>     * What rate(s) are you trying to record? There's a limit to how fast the ESP32 can record to SD. I haven't dug into it yet but 4Hz is currently the tested limit.

I've about the same limit with my own logger, a fork of https://github.com/PaulZC/ZED-F9P_FeatherWing_USB
I thinked that an esp32 could write more data to the micro-SD than a ATSAMD21G18, as there is more ram available.

nseidle, 2021-05-07:
I think this has been solved with v1.2. Please open a new issue if you need more assistance.

closed 2021-05-07 by nseidle