sparkfun/SparkFun_RTK_Firmware #12: Static base control Geographiv position input long and lat input reversed

Feinwerk, 2021-03-29:
Seems that the longitude and latitude input are saved reverse.

I am new to the RTK Surveyor got mine 2 this week i had no ECEF data available so i played around with geographical position input.
But i could not get it work with the rover. After reading the base setting out with u-center i could see that long and lat values are reversed. I entered them in reverse with ESP32 serial menu and now it seems to work.

Thank you

Alex

nseidle, 2021-03-29:
Hi Alex - thanks for reporting!

Yep, you're right. [Lat comes first](https://github.com/sparkfun/SparkFun_u-blox_GNSS_Arduino_Library/blob/master/src/SparkFun_u-blox_GNSS_Arduino_Library.cpp#L3591). Should be fixed in the next v1.2 release.

nseidle, 2021-05-07:
Fixed in v1.2. Thanks again for reporting.

closed 2021-05-07 by nseidle