sparkfun/SparkFun_RTK_Firmware #15: File date is 1 jan  1970

pyrog, 2021-05-05:
File `SFE_Surveyor_210423_134140.txt` was recorded probably with realease of Jan 7, 2021.

nseidle, 2021-05-05:
Which version of firmware are you using? You can find it by opening the serial menu.

We have a ton of good work being done on the release candidate for v1.2. File dates is [one of the fixes](https://github.com/sparkfun/SparkFun_RTK_Surveyor/commit/4221d4f828014b6633d62e6cfe541e57cee62172).

pyrog, 2021-05-05:
Unfortunately I tested it few weeks ago and I don't have it know.

But i'am happy to see that this issue is already solved 😄

pyrog, 2021-05-27:
Tested with 1.3 RC:
```
-rw-r--r-- 1 pyrog pyrog 1578864 mai   26 09:14 RTK_Surveyor_Firmware_v13RC_May25_combined.bin
-rw-r--r-- 1 pyrog pyrog 1578880 mai   26 13:52 RTK_Surveyor_Firmware_v13RC_May26_combined.bin
-rw-r--r-- 1 pyrog pyrog   20848 mai   27 09:18 SFE_Surveyor_210527_071717.ubx
-rw-r--r-- 1 pyrog pyrog    2647 mai   27 09:18 SFE_Surveyor_Settings.txt

```

nseidle, 2021-05-28:
Hi pyrog - just to be clear, the dates are showing correctly in your comment, correct?

nseidle, 2021-05-28:
I believe this is fixed in v1.3. Please re-open if it's still an issue.

closed 2021-05-28 by nseidle