sparkfun/SparkFun_RTK_Firmware #19: Add an NTRIP client in rover mode

pyrog, 2021-05-21:
Few and nice applications like [SW Maps](https://play.google.com/store/apps/details?id=np.com.softwel.swmaps&hl=fr&gl=US) have an "internal" NTRIP client to send RTCM data to the RTK Surveyor.
Other apps need to use i.e. [Lefebure NTRIP Client](https://play.google.com/store/apps/details?id=com.lefebure.ntripclient&hl=fr&gl=US).

With this feature, the RTK Surveyor could be more autonomous especially when used as a GNSS data logger.
It usage will be simpler on a smartphone, avoiding a third party app.

nseidle, 2021-05-24:
I don't quite understand your goal. You want to the Rover to be able to access RTCM data via NTRIP via SPP?

So rather than use Lefebure which uses the phone to hit a websocket and pass that RTCM data over SPP, you want the Surveyor to connect, over SPP, to a websocket, and get that RTCM data via NTRIP? I don't think that's possible. But perhaps I'm confused about the request.

Another way to get RTCM without a phone is to connect a cellular radio to the RADIO port of Surveyor. The cellular radio could easily connect to an NTRIP socket and pass the RTCM over serial to the Surveyor. This is effectively what the cell phone is doing, so I'm hesitant to design this radio because I don't want to have a 2nd monthly cell phone bill.

pyrog, 2021-05-24:
> You want to the Rover to be able to access RTCM data via NTRIP via SPP?

No, I want to use the WIFI for the RTK Surveyor in rover mode :smiley: 
It would be fine to use the [Serial Port Profile](https://learn.sparkfun.com/tutorials/bluetooth-basics/bluetooth-profiles) over bluetooth, but it require an app like Lefebure on the smartphone.

> This is effectively what the cell phone is doing, so I'm hesitant to design this radio because I don't want to have a 2nd monthly cell phone bill.

Me too :smiley: I just want to avoid a third app on the smartphone (or any app at all if we use the Surveyor as a data logger).
On some android smartphones, it's not obvious to enable the mock location.

On the field, I just check the RTK Surveyor Leds (RTK and Horizontal Accuracy).

pyrog, 2021-05-24:
> another way to get RTCM without a phone is to connect a cellular radio to the RADIO port of Surveyor. The cellular radio could easily connect to an NTRIP socket and pass the RTCM over serial to the Surveyor.

Or use an existing cellular modem over USB with an OTG cable?
i.e. [NOVA-R410 Global IoT Cellular USB Modem](https://www.sparkfun.com/products/15972)
But it need to write a "driver" to handle it over USB.

Or use it's UART?
(Nova exposes the u-blox modem's UART interface assolderablepads)

nseidle, 2021-05-25:
Your rover stays within WiFi range of a access point? I can understand it, but I've never heard of that.

After thinking about it, yes, I agree. This feature was not in the original because I thought rovers would be far outside of WiFi range but it should be an easy add. I really need to get a release candidate out for the other issues. I'll get to this one soon.

pyrog, 2021-05-26:
> Your rover stays within WiFi range of a access point? I can understand it, but I've never heard of that.
No, my rover stay in range of my cell phone that act as a WIFI access point :smiley: 

So I don't run any app in the cell phone, just share the "4G" connection.

nseidle, 2021-05-27:
And just to be clear, you are not using Bluetooth to connect a visualization/logging app like SW Maps? There may not be enough RAM left to have both WiFi and BT operating at the same time.

pyrog, 2021-05-28:
1. I use a BT app like SW Maps for some task,
2. I would like to use the RTK Surveyor "alone" for others.

I the second case, only the WIFI connection will be used.

This feature request is a bit too "early", so I close it.
One could reopen it if needed :smiley:

nseidle, 2021-05-28:
Let's keep it open so I don't forget about it. I think it's a fine addition to v1.4.

nseidle, 2021-11-30:
As an update, we've created a [NTRIP Client](https://github.com/sparkfun/SparkFun_u-blox_GNSS_Arduino_Library/tree/main/examples/ZED-F9P/Example15_NTRIPClient) so it is possible to implement a WiFi client on the RTK unit so that it shares your cellular connection to get RTCM data. I'm still thinking about how to cleanly implement the settings menus but it's doable. Let me know if this feature is still interesting. I think it's a lower priority but still a good one to add someday.

XPav, 2022-03-30:
This feature is interesting -- as an example, a vehicle with an on-board Wifi could use it.

nseidle, 2022-03-30:
Thanks @Xpav - I keep going putting this feature back on the shelf. The problem remains that the device would need simultaneous WiFi connection to the AP (car in this case) and BT to the phone (for viewing of the data). It's better to use the phone's cellular connection for SW Maps (viewer) and SW Maps (NTRIP) and have one wireless connection for simplicity, no?

mzandrew, 2022-03-30:
Hi Nate, Alex,

We could consider a different use case that does not involve an app on your
phone.  With just cell phone -> wifi sharing being active, you could roam
around with your rtk express and repeatedly hit the "mark" button and get a
set of ntrip-corrected location stamps on the microsd card for later
perusal.  Wouldn't need bluetooth in that use case, so may be more
straightforward to implement than you were thinking.

XPav, 2022-03-31:
I think it's a different Rover use case?  If you've got the Rover connected to Wifi, it think its likely because you want to share the NMEA with multiple devices on that network, a-la http://stripydog.blogspot.com/2015/03/nmea-0183-over-ip-unwritten-rules-for.html.  

UDP broadcast is of course pretty iffy over Wifi, but a TCP-server would work... although we're veering off into custom software land.

nseidle, 2022-04-04:
Ok folks. Give it a try. [Release Candidate](https://github.com/sparkfun/SparkFun_RTK_Firmware/releases/tag/v1.11) from Apr4 on will have a menu under GNSS to enable the NTRIP Client:

[image]

It needs *lots* of testing but it's generally working. GGA is correctly being pushed every 10 seconds. I added a 'Download' icon to indicate NTRIP Client. (I will add a similar upload icon from NTRIP Server mode soon)

It's scary how fast a ZED gets an RTK fix when the RTCM feed is good and near by (less than 10km).
[image]

nseidle, 2022-04-12:
Feature added and rolled into v1.11 release. Please open new issue if you find anything else.

closed 2022-04-12 by nseidle

pyrog, 2022-04-24:
Quick test, work fine. Thank you very much Nathan 😎

mzandrew, 2022-05-08:
Thanks for this feature.  It finally worked for me with version 1.12.  Before that, it refused to connect to my wifi?!?

BTW, for the use case I mentioned above (walk around a mark events with the "setup" button on the rtk express plus), after you get the ubx file, you can extract the marked points with the following:

grep -A1 CustomEvent SFE_Express_Plus_220507_192732.ubx | grep GPRMC > waypoints.nmea

And they import fine into google earth.