sparkfun/SparkFun_RTK_Firmware #20: startFixedBase() in Base.ino with ECEF failure

Darmann2, 2021-05-24:
In Base.ino function startFixedBase(): if you are working with ECEF, the call to setStaticPosition do not have the argument latLong(bool). Instead of that, the maxWait will be used as latLong.
So the ECEF values will be interpreted as Lat/Long/Alt.

` //Units are cm with a high precision extension so -1234.5678 should be called: (-123456, -78)`
`//-1280208.308,-4716803.847,4086665.811 is SparkFun HQ so...`
`response = i2cGNSS.setStaticPosition(majorEcefX, minorEcefX, majorEcefY, minorEcefY, majorEcefZ, minorEcefZ, maxWait);`

should be:
`response = i2cGNSS.setStaticPosition(majorEcefX, minorEcefX, majorEcefY, minorEcefY, majorEcefZ, minorEcefZ, false, maxWait);`

nseidle, 2021-05-25:
Well that's embarrassing! Good find and good fix. Thanks Darmann2! Will be in next RC.

nseidle, 2021-05-25:
Included in v13: [RTK_Surveyor_Firmware_v13RC_May25.zip](https://github.com/sparkfun/SparkFun_RTK_Surveyor/files/6542621/RTK_Surveyor_Firmware_v13RC_May25.zip)

closed 2021-05-25 by nseidle