sparkfun/SparkFun_RTK_Firmware #21: precision problem double to integer

Darmann2, 2021-05-25:
Hello!

I used a fixed position for the base station with ECEF format:
`3) Set ECEF X/Y/Z coordinates: 4226922.1216m, 1121263.0917m, 4628033.7264m`
Tracing the setStaticPosition position after a new power up bring the following arguments:
`setStaticPosition: (422692212 16) (112126309 17) (462803372 63) latLong=0 maxWait=2000`

So the Z coordinate is not the same as I have setup.

I changed the code of the startFixedBase() function in Base.ino to:
`long majorEcefX = floor((settings.fixedEcefX * 100.0) + 0.5);`
`long minorEcefX = floor((((settings.fixedEcefX * 100.0) - majorEcefX) * 100.0) + 0.5);`
`long majorEcefY = floor((settings.fixedEcefY * 100) + 0.5);`
`long minorEcefY = floor((((settings.fixedEcefY * 100.0) - majorEcefY) * 100.0) + 0.5);`
`long majorEcefZ = floor((settings.fixedEcefZ * 100) + 0.5);`
`long minorEcefZ = floor((((settings.fixedEcefZ * 100.0) - majorEcefZ) * 100.0) + 0.5);`

After that I got the following trace:
`setStaticPosition: (422692212 16) (112126309 17) (462803373 -36) latLong=0 maxWait=2000`

462803373 * 0.0001 + (-36 * 0.01) = 4628033,7264m

nseidle, 2021-05-25:
Thanks for reporting and good fix. I can confirm the problem. Will be in next RC shortly.

nseidle, 2021-05-25:
Please give [RTK_Surveyor_Firmware_v13RC_May25.zip](https://github.com/sparkfun/SparkFun_RTK_Surveyor/files/6542618/RTK_Surveyor_Firmware_v13RC_May25.zip) a try and report any issues.

closed 2021-05-25 by nseidle