sparkfun/SparkFun_RTK_Firmware #22: Can't upload firmware

pyrog, 2021-05-27:
I can't update the firmware using a SD card (see https://github.com/sparkfun/SparkFun_RTK_Surveyor/issues/18#issuecomment-848712207) so I tried to use esptool.

Under Linux/Ubuntu I use the python version of [esptool](https://github.com/espressif/esptool) : 
`esptool.py --chip esp32 --port /dev/ttyUSB0 --baud 921600 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 80m --flash_size detect 0x00 RTK_Surveyor_Firmware_v13RC_May26_combined.bin
`

```
esptool.py v3.0
Serial port /dev/ttyUSB0
Connecting....
Chip is ESP32-D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: ac:67:b2:1b:0c:e0
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...

A fatal error occurred: Timed out waiting for packet header
```

pyrog, 2021-05-27:
Trying a lower speed (**115200** is the actual speed of the RTK Surveyor) solve this issue :smiley: 

```
esptool.py --chip esp32 --port /dev/ttyUSB0 --baud 115200 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 80m --flash_size detect 0x00 RTK_Surveyor_Firmware_v13RC_May26_combined.bin 
esptool.py v3.0
Serial port /dev/ttyUSB1
Connecting......
Chip is ESP32-D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: ac:67:b2:1b:0c:e0
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 1578880 bytes to 905480...
Wrote 1578880 bytes (905480 compressed) at 0x00000000 in 79.9 seconds (effective 158.2 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```

closed 2021-05-27 by pyrog

nseidle, 2021-05-27:
Thanks for reporting your solution. 

>  (115200 is the actual speed of the RTK Surveyor) 

Just to be clear, the ESP32 does not have a set UART speed. This is negotiated during the start of bootload. I have however seen lots of issues with the higher UART speeds on *nix platforms. You've found the solution - lower the speed!

pyrog, 2021-05-29:
Note: I tried to update the firmware from microSD card from 1.3RC may 26 to 1.3RC may 28.
That worked fine :sunglasses: 

```
Menu: Update Firmware Menu
1) Load RTK_Surveyor_Firmware_v13RC_May28.bin

2) Load RTK_Surveyor_Firmware_v13RC_May25_combined.bin

3) Load RTK_Surveyor_Firmware_v13RC_May26_combined.bin

4) Load RTK_Surveyor_Firmware_v13RC_May28_combined.bin

x) Exit
1
WiFi Stopped
Bluetooth turned off
Loading RTK_Surveyor_Firmware_v13RC_May28.bin

[D][Updater.cpp:132] begin(): OTA Partition: app1
Moving file to OTA section
Bytes to write: 1514128
[=0%
[==5%
[===10%
[====15%
[=====20%
[======25%
[=======30%
[========35%
[=========40%
[==========45%
[===========50%
[============55%
[=============60%
[==============65%
[===============70%
[================75%
[=================80%
[==================85%
[===================90%
[====================95%
[=====================100%]
File move complete
Firmware updated successfully. Rebooting. Good bye!
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8
RTK Variant: SFE_Surveyor

Reset reason: ESP_RST_SW
microSD online
Unknown setting frequentFileAccessTimestamps on line: 0
MAX17048 configuration complete
Batt (98%): Voltage: 4.21V Charging: 7.49%/hr Green

getPortSettings failed
GNSS configuration complete
WiFi Stopped
[I][BluetoothSerial.cpp:572] _init_bt(): device name set
Bluetooth broadcasting as: Surveyor Rover-0CE2
State: Rover - No Fix
[I][BluetoothSerial.cpp:244] esp_spp_cb(): ESP_SPP_INIT_EVT
[I][BluetoothSerial.cpp:247] esp_spp_cb(): ESP_SPP_INIT_EVT: slave: start
[I][BluetoothSerial.cpp:348] esp_spp_cb(): ESP_SPP_START_EVT
Batt (98%): Voltage: 4.21V Charging: 7.49%/hr Green

Appending last available log
Log file created: SFE_Surveyor_210529_122623.ubx

UBX file size: 33 - Generation rate: 0.0kB/s - Write speed: infkB/s

```