sparkfun/SparkFun_RTK_Firmware #24: FR: Enable/Disable message group

Stefal, 2021-05-29:
### Subject of the issue
With the latest RC, if I want to enable/disable ubx messages or nmea messages logging, I have to change each message value to 0/1. It's a long process. And when I need to enable back Nmea, I don't remember the default settings, so I end up with a complete reset.

### Your workbench
* What version of RTK Surveyor firmware are you running? 
13RC_May28

### Expected behavior
Inside logging menu we could have a on/off setting for nmea and ubx, like the previous releases

pyrog, 2021-05-29:
> It's a long process.

That's what I mean in in https://github.com/sparkfun/SparkFun_RTK_Surveyor/issues/18#issuecomment-850216333_
- Setting the logging is a bit tedious.

nseidle, 2021-06-01:
Let's see so a menu item that was something like:

* "Set to Surveying defaults" - clears all msg, enables the 5 NMEA messages GGA, GSA, GST, GSV, and RMC
* "Set to PPP Logging defaults" - clears all msg, enables the 5 NMEA messages GGA, GSA, GST, GSV, and RMC, as well as RAWX and SFRBX

How does that sound?

pyrog, 2021-06-03:
> How does that sound?
Sound good :smiley: 

I have a suggestion for the Surveyor that is also usable for the [RTK Express](https://www.sparkfun.com/products/18019)...
Have a user editable list of settings and for the RTK Surveyor:
Select 1 setting of the list for the base mode, and another 1 setting for the rover mode.

For the RTK Express, select 1 of the settings of the list.

Examples:
- "SW maps hiking" (NMEA at 1 Hz + DYN_MODEL_PEDESTRIAN)
- "SW maps car" (NMEA  at 4 Hz + DYN_MODEL_AUTOMOTIVE)
- "Lefebure biking" (UBX  at 2 Hz + DYN_MODEL_BIKE)
- "Drone" (UBX at 10 Hz + DYN_MODEL_AIRBORNE1g)

One could set the Surveyor with _"SW maps car"_  in Rover Setup, and  _"SW maps hiking"_ in Base Setup.

nseidle, 2021-06-07:
@pyrog - Oh how the features creep ;) What you're asking for is effectively multiple configuration files and being able to select between them. Ok, maybe. But let's move that to a new issue if you want to.

@Stefal - I've added the ability to reset to two defaults as well as all off, and all messages on (useful for my testing).

[image]

This will be in next RC.

closed 2021-06-07 by nseidle