sparkfun/SparkFun_RTK_Firmware #25: Crash during logging

Stefal, 2021-05-30:
### Subject of the issue
The RTK Surveyor logging crashes

### Your workbench
* What version of RTK Surveyor firmware are you running? 
13RC_May28

### Steps to reproduce
Enable RAWX, SFRBX, HPPOSLLH, HPPOSECEF, DOP, PVT, SAT, SIG, TIMEGPS, TIMEUTC

### Actual behavior
After a few minutes, the RTK Surveyor crash and reboot

```
UBX file size: 543394 - Generation rate: 14.5kB/s - Write speed: 3768.4kB/s
Rover Accuracy (m): 9.9342
Rover Accuracy (m): 9.3130
ets Jun  8 2016 00:22:57

rst:0x8 (TG1WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8
RTK Variant: SFE_Surveyor
Reset reason: ESP_RST_INT_WDT
Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0)
Core 0 register dump:
PC      : 0x40081b71  PS      : 0x00050034  A0      : 0x4008e548  A1      : 0x3ffc0df0
A2      : 0x00000000  A3      : 0x00000003  A4      : 0x00000001  A5      : 0x40093948
A6      : 0x00000000  A7      : 0x3ffbd8b4  A8      : 0x00000000  A9      : 0x3ff40000
A10     : 0x00000180  A11     : 0x00000180  A12     : 0x00000000  A13     : 0x3ffbafd0
A14     : 0x3ffb9a20  A15     : 0xbaad5678  SAR     : 0x0000001e  EXCCAUSE: 0x00000005
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000
Core 0 was running in ISR context:
EPC1    : 0x4000bff0  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x40081b71

ELF file SHA256: 0000000000000000

Backtrace: 0x40081b71:0x3ffc0df0 0x4008e545:0x3ffc0e20 0x4000bfed:0x3ffbb020 0x400923ad:0x3ffbb030 0x401b8f12:0x3ffbb050 0x401b8f6f:0x3ffbb090 0x400ea820:0x3ffbb0c0 0x400ea8e7:0x3ffbb0e0 0x400eae1d:0x3ffbb100 0x400eb2bd:0x3ffbb140 0x400d2100:0x3ffbb180 0x4009122a:0x3ffbb1b0

Core 1 register dump:
PC      : 0x401e084a  PS      : 0x00060734  A0      : 0x801b84f2  A1      : 0x3ffbc680
A2      : 0x00000000  A3      : 0x80000001  A4      : 0x00000000  A5      : 0x00000001
A6      : 0x00060a20  A7      : 0x00000000  A8      : 0x800923b0  A9      : 0x3ffbc670
A10     : 0x00000003  A11     : 0x00060a23  A12     : 0x00060a20  A13     : 0x3ffbd680
A14     : 0x401b96a8  A15     : 0x00000000  SAR     : 0x00000000  EXCCAUSE: 0x00000005
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000

ELF file SHA256: 0000000000000000

Backtrace: 0x401e084a:0x3ffbc680 0x401b84ef:0x3ffbc6a0 0x400929e9:0x3ffbc6c0 0x4009122a:0x3ffbc6e0

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8
RTK Variant: SFE_Surveyor
Reset reason: ESP_RST_INT_WDT
microSD online
MAX17048 configuration complete
Batt (28%): Voltage: 3.99V Charging: 17.06%/hr Yellow
GNSS configuration complete
WiFi Stopped
[I][BluetoothSerial.cpp:572] _init_bt(): device name set
Bluetooth broadcasting as: Surveyor Rover-0CEE
State: Rover - No Fix
[I][BluetoothSerial.cpp:244] esp_spp_cb(): ESP_SPP_INIT_EVT
[I][BluetoothSerial.cpp:247] esp_spp_cb(): ESP_SPP_INIT_EVT: slave: start
[I][BluetoothSerial.cpp:348] esp_spp_cb(): ESP_SPP_START_EVT
```

```
UBX file size: 6469659 - Generation rate: 17.5kB/s - Write speed: 4932.9kB/s
Rover Accuracy (m): 2.3703
ets Jun  8 2016 00:22:57

rst:0x8 (TG1WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8
RTK Variant: SFE_Surveyor
Reset reason: ESP_RST_INT_WDT
microSD online
MAX17048 configuration complete
Batt (43%): Voltage: 4.14V Charging: 98.59%/hr Yellow
getPortSettings failed
getSerialRate failed
Updating UART1 rate
getSerialRate failed
Updating UART2 rate
Module failed initial config.
Module failed to save.
Failed to configure module. Trying again.
getSerialRate failed
Updating UART2 rate
GNSS configuration complete
WiFi Stopped
[I][BluetoothSerial.cpp:572] _init_bt(): device name set
Bluetooth broadcasting as: Surveyor Rover-0CEE
State: Rover - No Fix
[I][BluetoothSerial.cpp:244] esp_spp_cb(): ESP_SPP_INIT_EVT
[I][BluetoothSerial.cpp:247] esp_spp_cb(): ESP_SPP_INIT_EVT: slave: start
[I][BluetoothSerial.cpp:348] esp_spp_cb(): ESP_SPP_START_EVT
Batt (43%): Voltage: 4.14V Charging: 98.59%/hr Yellow
System time set to: May 30 2021 08:28:21
Appending last available log
Log file created: SFE_Surveyor_210528_164857.ubx
UBX file size: 153287796 - Generation rate: 30657.6kB/s - Write speed: infkB/s
State: Rover - Fix
Rover Accuracy (m): 3.4061
Rover Accuracy (m): 3.2876
Rover Accuracy (m): 3.1985
Batt (43%): Voltage: 4.14V Charging: 98.59%/hr Yellow
UBX file size: 153371437 - Generation rate: 16.7kB/s - Write speed: 19303.3kB/s
Rover Accuracy (m): 3.1209
```

nseidle, 2021-06-01:
Hi Stefal - My apologies. I think something may be wrong with the v13RC-May28 binaries. Would you please give the [May31  binaries](https://github.com/sparkfun/SparkFun_RTK_Surveyor/releases/tag/v1.3) a try? These seem to have resolve all of pyrog's [issues](https://github.com/sparkfun/SparkFun_RTK_Surveyor/issues/26).

I will be running some longer, multi-hour logging tests as well to try to replicate your issue.

Stefal, 2021-06-02:
Yes, that's better now, I can log, but I still see some errors:
```
[E][BluetoothSerial.cpp:317] esp_spp_cb(): RX Full! Discarding 75 bytes
```

```
UBX file size: 10358866 - Generation rate: 14.2kB/s - Write speed: 9213.0kB/s
[E][BluetoothSerial.cpp:182] _spp_send_buffer(): SPP Write Congested!
[E][BluetoothSerial.cpp:182] _spp_send_buffer(): SPP Write Congested!
Rover Accuracy (m): 0.0141
Rover Accuracy (m): 0.0141
Batt (91%): Voltage: 4.19V Charging: 24.34%/hr Green
```

```
[E][BluetoothSerial.cpp:182] _spp_send_buffer(): SPP Write Congested!
Rover Accuracy (m): 0.0141
[E][BluetoothSerial.cpp:146] _spp_queue_packet(): SPP TX Packet Malloc Failed!
[E][BluetoothSerial.cpp:146] _spp_queue_packet(): SPP TX Packet Malloc Failed!
Rover Accuracy (m): 0.0141
[E][BluetoothSerial.cpp:146] _spp_queue_packet(): SPP TX Packet Malloc Failed!
Batt (91%): Voltage: 4.19V Charging: 24.34%/hr Green
UBX file size: 10805150 - Generation rate: 15.1kB/s - Write speed: 6780.2kB/s
Rover Accuracy (m): 0.0141
[E][BluetoothSerial.cpp:146] _spp_queue_packet(): SPP TX Packet Malloc Failed!
[E][BluetoothSerial.cpp:146] _spp_queue_packet(): SPP TX Packet Malloc Failed!
[E][BluetoothSerial.cpp:146] _spp_queue_packet(): SPP TX Packet Malloc Failed!
Rover Accuracy (m): 0.0141
[E][BluetoothSerial.cpp:146] _spp_queue_packet(): SPP TX Packet Malloc Failed!
[E][BluetoothSerial.cpp:146] _spp_queue_packet(): SPP TX Packet Malloc Failed!
[E][BluetoothSerial.cpp:146] _spp_queue_packet(): SPP TX Packet Malloc Failed!
[E][BluetoothSerial.cpp:146] _spp_queue_packet(): SPP TX Packet Malloc Failed!
Guru Meditation Error: Core  0 panic'ed (StoreProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x4016ba46  PS      : 0x00060430  A0      : 0x801288d8  A1      : 0x3ffdc2f0
A2      : 0x3ffffb4c  A3      : 0x00000081  A4      : 0x000003eb  A5      : 0x3ffffb54
A6      : 0x00000000  A7      : 0x00001100  A8      : 0x8016ba41  A9      : 0x3ffdc2a0
A10     : 0x00000000  A11     : 0x000003f3  A12     : 0x00000000  A13     : 0xffffffff
A14     : 0x00001800  A15     : 0x00000000  SAR     : 0x00000010  EXCCAUSE: 0x0000001d
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000

ELF file SHA256: 0000000000000000

Backtrace: 0x4016ba46:0x3ffdc2f0 0x401288d5:0x3ffdc310 0x4016b7d5:0x3ffdc330 0x4016b7e8:0x3ffdc350 0x40156fff:0x3ffdc370 0x4016b50d:0x3ffdc390 0x4009122a:0x3ffdc3c0

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8
SparkFun RTK Surveyor v1.3-May 31 2021
Reset reason: ESP_RST_PANIC
microSD online
Unknown setting rtkFirmwareVersion on line: v1.3-May 31 2021
MAX17048 configuration complete
Batt (91%): Voltage: 4.19V Charging: 24.34%/hr Green
getPortSettings failed
getPortSettings failed
GNSS configuration complete
```

Could it be related to the Surveyor trying to send too much ubx messages to the bluetooth "output"? I think that this feature should be  an option and disabled as default. Or a separate ubx message selection for logging and bluetooth output... but it starts to become complex 🤔

nseidle, 2021-06-03:
I have the same messages enabled and see the following:

> 21:12:03.895 -> Rover Accuracy (m): 0.1448
> 21:12:05.803 -> Batt (86%): Voltage: 4.08V Charging: 2.29%/hr Green
> 21:12:05.803 -> 
> Rover Accuracy (m): 0.1449
> 21:12:06.215 -> UBX file size: 5382144 - Generation rate: 34.4kB/s - Write speed: 12982.6kB/s
> 21:12:07.993 -> Rover Accuracy (m): 0.1450
> 21:12:10.116 -> Rover Accuracy (m): 0.1452
> 21:12:10.817 -> Batt (86%): Voltage: 4.08V Charging: 2.29%/hr Green
> 21:12:10.817 -> 
> UBX file size: 5550080 - Generation rate: 33.6kB/s - Write speed: 10367.1kB/s
> 21:12:12.209 -> Rover Accuracy (m): 0.1452
> 21:12:14.222 -> Rover Accuracy (m): 0.1453
> 21:12:15.846 -> Batt (86%): Voltage: 4.08V Charging: 2.29%/hr Green
> 21:12:15.846 -> 
> UBX file size: 5722112 - Generation rate: 34.4kB/s - Write speed: 9078.7kB/s
> 21:12:16.265 -> Rover Accuracy (m): 0.1453

My generation rate is much larger (I'll look into that). I have a "[slow](https://www.sparkfun.com/products/15107)" SD card inserted in an attempt to slow the record speed. My theory is that a slow SD card needs more time to record, and because SD recording is part of SPP writing, a slow SD write can cause the SPP broadcast to slow down.

Ahah! - A reboot:

> 21:13:48.213 -> Rover Accuracy (m): 0.1474
> 21:13:49.985 -> ets Jul 29 2019 12:21:46
> 21:13:49.985 -> 
> 21:13:49.985 -> rst:0x8 (TG1WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
> 21:13:49.985 -> configsip: 0, SPIWP:0xee
> 21:13:49.985 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
> 21:13:49.985 -> mode:DIO, clock div:1
> 21:13:49.985 -> load:0x3fff0018,len:4
> 21:13:49.985 -> load:0x3fff001c,len:1216
> 21:13:49.985 -> ho 0 tail 12 room 4
> 21:13:49.985 -> load:0x40078000,len:10944
> 21:13:49.985 -> load:0x40080400,len:6388
> 21:13:49.985 -> entry 0x400806b4
> 21:13:50.437 -> SparkFun RTK Express v1.3-Jun  2 2021
> 21:13:50.437 -> Reset reason: ESP_RST_INT_WDT
> 21:13:50.810 -> Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0)
> 21:13:50.810 -> Core 0 register dump:
> 21:13:50.810 -> PC      : 0x40081b51  PS      : 0x00050034  A0      : 0x4008e548  A1      : 0x3ffc0df0  
> 21:13:50.810 -> A2      : 0x3ffbedd8  A3      : 0x00000000  A4      : 0x00000001  A5      : 0x40093948  
> 21:13:50.810 -> A6      : 0x00000000  A7      : 0x3ffbd8b4  A8      : 0x00000000  A9      : 0x3ff40000  
> 21:13:50.856 -> A10     : 0x00000000  A11     : 0x00300600  A12     : 0x00000000  A13     : 0x3ffbafd0  
> 21:13:50.856 -> A14     : 0x3ffb9a20  A15     : 0xbaad5678  SAR     : 0x0000001e  EXCCAUSE: 0x00000005  
> 21:13:50.856 -> EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  
> 21:13:50.856 -> Core 0 was running in ISR context:
> 21:13:50.856 -> EPC1    : 0x4000bff0  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x40081b51
> 21:13:50.856 -> 
> 21:13:50.856 -> ELF file SHA256: 0000000000000000
> 21:13:50.856 -> 
> 21:13:50.856 -> Backtrace: 0x40081b51:0x3ffc0df0 0x4008e545:0x3ffc0e20 0x4000bfed:0x3ffbb020 0x400923ad:0x3ffbb030 0x401b9012:0x3ffbb050 0x401b906f:0x3ffbb090 0x400ea920:0x3ffbb0c0 0x400ea9e7:0x3ffbb0e0 0x400eaf1d:0x3ffbb100 0x400eb3bd:0x3ffbb140 0x400d2118:0x3ffbb180 0x4009122a:0x3ffbb1b0
> 21:13:50.903 -> 
> 21:13:50.903 -> Core 1 register dump:
> 21:13:50.903 -> PC      : 0x401e094a  PS      : 0x00060934  A0      : 0x801b85f2  A1      : 0x3ffbc680  
> 21:13:50.903 -> A2      : 0x00000000  A3      : 0x80000001  A4      : 0x00000000  A5      : 0x00000001  
> 21:13:50.903 -> A6      : 0x00060a20  A7      : 0x00000000  A8      : 0x800923b0  A9      : 0x3ffbc670  
> 21:13:50.903 -> A10     : 0x00000003  A11     : 0x00060a23  A12     : 0x00060a20  A13     : 0x3ffbd680  
> 21:13:50.951 -> A14     : 0x401b97a8  A15     : 0x00000000  SAR     : 0x00000000  EXCCAUSE: 0x00000005  
> 21:13:50.951 -> EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  
> 21:13:50.951 -> 
> 21:13:50.951 -> ELF file SHA256: 0000000000000000
> 21:13:50.951 -> 
> 21:13:50.951 -> Backtrace: 0x401e094a:0x3ffbc680 0x401b85ef:0x3ffbc6a0 0x400929e9:0x3ffbc6c0 0x4009122a:0x3ffbc6e0
> 21:13:50.951 -> 
> 21:13:50.951 -> Rebooting...
> 21:13:50.951 -> ets Jul 29 2019 12:21:46
> 21:13:50.951 -> 
> 21:13:50.951 -> rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
> 21:13:50.951 -> configsip: 0, SPIWP:0xee
> 21:13:50.951 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
> 21:13:50.998 -> mode:DIO, clock div:1
> 21:13:50.998 -> load:0x3fff0018,len:4
> 21:13:50.998 -> load:0x3fff001c,len:1216
> 21:13:50.998 -> ho 0 tail 12 room 4
> 21:13:50.998 -> load:0x40078000,len:10944
> 21:13:50.998 -> load:0x40080400,len:6388
> 21:13:50.998 -> entry 0x400806b4
> 21:13:51.421 -> SparkFun RTK Express v1.3-Jun  2 2021
> 21:13:51.421 -> Reset reason: ESP_RST_INT_WDT
> 21:13:51.562 -> microSD online
> 21:13:51.785 -> MAX17048 configuration complete
> 21:13:51.785 -> Batt (86%): Voltage: 4.08V Charging: 1.04%/hr Green
> 21:13:51.785 -> GNSS configuration complete
 
Tracing the stack shows us the below:

[image]

UART/SPP related. Curiously I don't see the same 

> [E][BluetoothSerial.cpp:146] _spp_queue_packet(): SPP TX Packet Malloc Failed!

but I can cause system reboot with BT connect. I'll try increasing the TX buffer.

pyrog, 2021-06-03:
BlueTooth disconnected yesterday morning during a 1.5 hours trip.
The RTK Surveyor need to be rebooted manually :thinking: 

- Lefebure NTRIP client connected in BT.
- No USB logging of the ESP32.
[image]

In the evening, same configuration + USB logging: no crash

nseidle, 2021-06-04:
@stefal - Would you please post your settings file? (You can drag/drop into github comment) I am having difficulty replicating.

> 13:08:06.260 -> Rover Accuracy (m): 0.3660
> 13:08:08.373 -> Rover Accuracy (m): 0.3666
> 13:08:10.382 -> Rover Accuracy (m): 0.3671
> 13:08:10.757 -> Batt (100%): Voltage: 4.19V Discharging: 0.00%/hr Green
> 13:08:10.757 -> 
> UBX file size: 257467957 - Generation rate: 21.6kB/s - Write speed: 18535.1kB/s
> 13:08:12.390 -> Rover Accuracy (m): 0.3677
> 13:08:14.451 -> Rover Accuracy (m): 0.3680
> 13:08:15.758 -> Batt (100%): Voltage: 4.19V Discharging: 0.00%/hr Green
> 13:08:15.758 -> 
> UBX file size: 257575057 - Generation rate: 21.4kB/s - Write speed: 23853.0kB/s

I am at 4Hz with your msgs on, connected to SW Maps, and have logged for a few hours with no resets. I am using 16GB Kingston card.

nseidle, 2021-06-04:
I do enjoy this game. I keep pushing the ESP32 and it keeps delivering! Increasing the report rate to 6Hz, then 8Hz is an obvious way to increase generation rate. However, at 10Hz we overwhelm the UART1 because it is limited to 481600bps at the moment. 

> 13:57:40.342 -> Batt (101%): Voltage: 4.20V Charging: 0.62%/hr Green
> 13:57:40.342 -> 
> UBX file size: 8084338 - Generation rate: 32.8kB/s - Write speed: 16497.8kB/s
> 13:57:40.998 -> Rover Accuracy (m): 0.0141
> 13:57:43.059 -> Rover Accuracy (m): 0.0141
> 13:57:45.164 -> Rover Accuracy (m): 0.0141
> 13:57:45.351 -> Batt (101%): Voltage: 4.20V Charging: 0.62%/hr Green
> 13:57:45.351 -> 
> UBX file size: 8252274 - Generation rate: 33.6kB/s - Write speed: 23376.4kB/s
> 13:57:47.263 -> Rover Accuracy (m): 0.0141
> 13:57:49.417 -> Rover Accuracy (m): 0.0141

Above: Recording just fine without resets, frustratingly.

We need to isolate how/when the resets are occurring.

* Do we see resets with no microSD inserted?
* Do we see resets with no Bluetooth connection?
* Is there a generation rate that once we go above leads to resets?

My current theory is that a misbehaving SD card, combined with a large amount of SPP data to transmit can cause problems. I have exposed the size of the TX and RX SPP queues in the latest push (v13RC_June4). Please give it a try. I cannot yet reliably reset a unit with an TX buffer of 32 (the original size) so I can't say if increasing it helps yet.

nseidle, 2021-06-04:
41k generation rate at 6.67Hz, RX/TX SPP Buffer is 2048/32, SD card is 1GB SparkX (cheap):

> 14:31:18.005 -> UBX file size: 14267674 - Generation rate: 41.8kB/s - Write speed: 35911.3kB/s
> 14:31:18.334 -> Task freeHeap: 103880
> 14:31:18.334 -> freeHeap: 103472
> 14:31:18.568 -> Task freeHeap: 103880
> 14:31:19.365 -> freeHeap: 103876
> 14:31:19.599 -> Rover Accuracy (m): 0.2587

I am seeing dropped characters inside the log, so that's a separate issue, but the device doesn't seem to want to fall over.

pyrog, 2021-06-04:
Got a lot of crash today during a 10 hours trip.
64 GB PNY Elite microSD 
BT connection to Lefebure NTRIP Client
```
sizeOfSettings=2576
rtkIdentifier=19
rtkFirmwareVersion=v1.3-May 31 2021
printDebugMessages=0
enableSD=1
enableDisplay=1
maxLogTime_minutes=600
observationSeconds=60
observationPositionAccuracy=5.00
fixedBase=0
fixedBaseCoordinateType=0
fixedEcefX=0.00
fixedEcefY=0.00
fixedEcefZ=0.00
fixedLat=0.000000000
fixedLong=0.000000000
fixedAltitude=0.0000
dataPortBaud=460800
radioPortBaud=57600
enableSBAS=0
enableNtripServer=0
casterHost=caster.centipede.fr
casterPort=2101
mountPoint=bldr_dwntwn2
mountPointPW=Surveyor 0CE2
wifiSSID=TRex
wifiPW=parachutes
surveyInStartingAccuracy=1.00
measurementRate=250
navigationRate=1
enableI2Cdebug=0
enableHeapReport=0
dataPortChannel=0
spiFrequency=8
enableLogging=1
message.nmea_dtm.msgRate=0
message.nmea_gbs.msgRate=0
message.nmea_gga.msgRate=1
message.nmea_gll.msgRate=0
message.nmea_gns.msgRate=0
message.nmea_grs.msgRate=0
message.nmea_gsa.msgRate=1
message.nmea_gst.msgRate=1
message.nmea_gsv.msgRate=1
message.nmea_rmc.msgRate=1
message.nmea_vlw.msgRate=0
message.nmea_vtg.msgRate=0
message.nmea_zda.msgRate=0
message.nav_clock.msgRate=0
message.nav_dop.msgRate=1
message.nav_eoe.msgRate=0
message.nav_geofence.msgRate=0
message.nav_hpposecef.msgRate=0
message.nav_hpposllh.msgRate=1
message.nav_odo.msgRate=0
message.nav_orb.msgRate=0
message.nav_posecef.msgRate=0
message.nav_posllh.msgRate=0
message.nav_pvt.msgRate=1
message.nav_relposned.msgRate=0
message.nav_sat.msgRate=1
message.nav_sig.msgRate=1
message.nav_status.msgRate=0
message.nav_svin.msgRate=0
message.nav_timebds.msgRate=0
message.nav_timegal.msgRate=0
message.nav_timeglo.msgRate=0
message.nav_timegps.msgRate=1
message.nav_timels.msgRate=0
message.nav_timeutc.msgRate=0
message.nav_velecef.msgRate=0
message.nav_velned.msgRate=0
message.rxm_measx.msgRate=0
message.rxm_rawx.msgRate=1
message.rxm_rlm.msgRate=0
message.rxm_rtcm.msgRate=0
message.rxm_sfrbx.msgRate=1
message.mon_comms.msgRate=0
message.mon_hw2.msgRate=0
message.mon_hw3.msgRate=0
message.mon_hw.msgRate=0
message.mon_io.msgRate=0
message.mon_msgpp.msgRate=0
message.mon_rf.msgRate=0
message.mon_rxbuf.msgRate=0
message.mon_rxr.msgRate=0
message.mon_txbuf.msgRate=0
message.tim_tm2.msgRate=0
message.tim_tp.msgRate=0
message.tim_vrfy.msgRate=0
message.rtcm_1005.msgRate=0
message.rtcm_1074.msgRate=0
message.rtcm_1077.msgRate=0
message.rtcm_1084.msgRate=0
message.rtcm_1087.msgRate=0
message.rtcm_1094.msgRate=0
message.rtcm_1097.msgRate=0
message.rtcm_1124.msgRate=0
message.rtcm_1127.msgRate=0
message.rtcm_1230.msgRate=0
message.rtcm_4072_0.msgRate=0
message.rtcm_4072_1.msgRate=0
```

[image]
[image]

RTK Surveyor must be manually rebooted

Sorry for the short report, I am in holidays without any laptop

Stefal, 2021-06-11:
> @Stefal - Would you please post your settings file? (You can drag/drop into github comment) I am having difficulty replicating.

Sorry to answer so late, I'm busy.

I don't remember my settings. I'll try to make some more tests this week-end.

pyrog, 2021-06-14:
After firmware upgrade to v1.3-Jun 4 2021, I don't observe any crash 😃
But data loss see #37

nseidle, 2021-06-22:
I'm going to close this due to inactivity. I think we've improved the stability considerably. If you have any new reports by all means please open a new issue.

closed 2021-06-22 by nseidle