sparkfun/SparkFun_RTK_Firmware #26: Stall during logging

pyrog, 2021-05-31:
### Subject of the issue
- BlueTooth disconnection.
- Logging stopped during 13 seconds.
See also https://github.com/sparkfun/SparkFun_RTK_Surveyor/issues/18#issuecomment-850512116.

### Your workbench
*  v13RC May 28.
* BT connexion to [SW Maps](https://play.google.com/store/apps/details?id=np.com.softwel.swmaps&hl=fr&gl=US) and [Lefebure NTRIP Client](https://play.google.com/store/apps/details?id=com.lefebure.ntripclient&hl=fr&gl=US).
*  [Default settings](https://github.com/sparkfun/SparkFun_RTK_Surveyor/files/6568277/SFE_Surveyor_Settings.txt) (log only NMEA messages).

### Actual behavior
* Forward trip:
  * 64 mn
  * 3 disconnections (12s, 11s, 18s)
* Return trip:
  * 53 mn
  * 4 disconnections (11s, 13s, 14s, 15s)

#### Disconnection
[image]
[image]

#### Data loss
[image]
- grey square: gps fix missing
- yellow cross: no data at all

pyrog, 2021-05-31:
I also made a quick test without microSD card.
It seam that the RTK Surveyor don't close the BT connexion.

pyrog, 2021-05-31:
New test [SFE_Surveyor_Settings.txt](https://github.com/sparkfun/SparkFun_RTK_Surveyor/files/6570160/SFE_Surveyor_Settings.txt):

- microSD card inserted
- Lefebure NTRIP client connected over
- NMEA messages disabled
- UBX messages enabled (nav_dop, nav_hpposllh, nav_pvt, nav_sat, nav_sig, nav_timegps, rxm_sfrbx)

Normal operation
```
...
10:24:39.041 UBX file size: 4144320 - Generation rate: 11.7kB/s - Write speed: 8352.4kB/s
10:24:39.046 Rover Accuracy (m): 0.0374
10:24:39.062 Rover Accuracy (m): 0.0331
10:24:41.083 Rover Accuracy (m): 0.0368
10:24:43.096 Batt (98%): Voltage: 4.21V Charging: 7.28%/hr Green
10:24:44.043 UBX file size: 4203040 - Generation rate: 11.7kB/s - Write speed: 10928.7kB/s
10:24:44.049 Rover Accuracy (m): 0.0377
10:24:45.182 Rover Accuracy (m): 0.0344
10:24:47.215 Batt (98%): Voltage: 4.21V Charging: 7.28%/hr Green
10:24:49.051 UBX file size: 4261616 - Generation rate: 11.7kB/s - Write speed: 3320.1kB/s
10:24:49.056 Rover Accuracy (m): 0.0322
10:24:49.230 Rover Accuracy (m): 0.0329
10:24:51.252 Rover Accuracy (m): 0.0409
```
Crash
```
10:24:53.274 ets Jun  8 2016 00:22:57
10:24:54.259 
10:24:54.259 rst:0x8 (TG1WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
10:24:54.264 configsip: 0, SPIWP:0xee
10:24:54.267 clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
10:24:54.274 mode:DIO, clock div:1
10:24:54.276 load:0x3fff0018,len:4
10:24:54.278 load:0x3fff001c,len:1216
10:24:54.282 ho 0 tail 12 room 4
10:24:54.282 load:0x40078000,len:9720
10:24:54.284 ho 0 tail 12 room 4
10:24:54.287 load:0x40080400,len:6352
10:24:54.290 entry 0x400806b8
10:24:54.290 RTK Variant: SFE_Surveyor
10:24:54.734 Reset reason: ESP_RST_INT_WDT
10:24:54.737 microSD online
10:24:54.800 MAX17048 configuration complete
10:24:54.920 Batt (98%): Voltage: 4.19V Charging: 5.41%/hr Green
10:24:55.954 getPortSettings failed
10:24:55.954 getPortSettings failed
10:24:57.109 Module failed initial config.
10:24:58.715 Module failed to save.
10:24:58.970 Failed to configure module. Trying again.
10:24:58.974 getPortSettings failed
10:25:00.583 getPortSettings failed
10:25:01.401 getPortSettings failed
10:25:02.172 GNSS configuration complete
10:25:06.177 WiFi Stopped
10:25:08.696 [I][BluetoothSerial.cpp:572] _init_bt(): device name set
10:25:09.204 Bluetooth broadcasting as: Surveyor Rover-0CE2
10:25:09.209 State: Rover - No Fix
10:25:09.212 [I][BluetoothSerial.cpp:244] esp_spp_cb(): ESP_SPP_INIT_EVT
10:25:09.215 [I][BluetoothSerial.cpp:247] esp_spp_cb(): ESP_SPP_INIT_EVT: slave: start
10:25:09.223 [I][BluetoothSerial.cpp:348] esp_spp_cb(): ESP_SPP_START_EVT
10:25:09.229 Batt (98%): Voltage: 4.19V Charging: 5.41%/hr Green
10:25:09.231 System time set to: May 31 2021 08:25:10
10:25:09.236 Appending last available log
10:25:09.351 Log file created: SFE_Surveyor_210531_081835.ubx
10:25:09.359 UBX file size: 4273617 - Generation rate: 854.7kB/s - Write speed: infkB/s
10:25:09.364 State: Rover - Fix
10:25:09.367 Rover Accuracy (m): 0.0634
10:25:09.890 State: Rover - RTK Float
10:25:09.893 [I][BluetoothSerial.cpp:255] esp_spp_cb(): ESP_SPP_SRV_OPEN_EVT: 0
10:25:10.058 Client Connected
10:25:10.058 Rover Accuracy (m): 0.0410
10:25:11.958 Rover Accuracy (m): 0.0400
10:25:13.967 Batt (98%): Voltage: 4.19V Charging: 5.41%/hr Green
10:25:14.358 UBX file size: 4326287 - Generation rate: 10.5kB/s - Write speed: 3056.3kB/s
```

[image]

See #25

pyrog, 2021-05-31:
Another test with same settings [SFE_Surveyor_Settings.txt](https://github.com/sparkfun/SparkFun_RTK_Surveyor/files/6570160/SFE_Surveyor_Settings.txt):

- microSD card **removed**
- Lefebure NTRIP client connected over Blue Tooth
- Serial terminal connected over USB
- NMEA messages disabled
- UBX messages enabled (nav_dop, nav_hpposllh, nav_pvt, nav_sat, nav_sig, nav_timegps, rxm_sfrbx)

[image] [image]

First crash around 11:34:52
```
11:34:52.204 [E][BluetoothSerial.cpp:317] esp_spp_cb(): RX Full! Discarding 97 bytes
11:34:52.209 ^J[E][BluetoothSerial.cpp:317] esp_spp_cb(): RX Full! Discarding 700 bytes
11:34:52.215 ^J[E][BluetoothSerial.cpp:317] esp_spp_cb(): RX Full! Discarding 470 bytes
11:34:52.220 ^J[E][BluetoothSerial.cpp:317] esp_spp_cb(): RX Full! Discarding 990 bytes
11:34:52.229 ^J[E][BluetoothSerial.cpp:317] esp_spp_cb(): RX Full! Discarding 382 bytes
11:34:52.234 ^J[E][BluetoothSerial.cpp:317] esp_spp_cb(): RX Full! Discarding 990 bytes
11:34:52.240 ^J[E][BluetoothSerial.cpp:317] esp_spp_cb(): RX Full! Discarding 545 bytes
11:34:52.247 ^JRover Accuracy (m): 0.0338
11:34:52.499 ^Jets Jun  8 2016 00:22:57
11:34:54.447 ^J
11:34:54.447 ^Jrst:0x8 (TG1WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
11:34:54.452 ^Jconfigsip: 0, SPIWP:0xee
11:34:54.455 ^Jclk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
11:34:54.461 ^Jmode:DIO, clock div:1
11:34:54.463 ^Jload:0x3fff0018,len:4
11:34:54.466 ^Jload:0x3fff001c,len:1216
11:34:54.469 ^Jho 0 tail 12 room 4
11:34:54.469 ^Jload:0x40078000,len:9720
11:34:54.472 ^Jho 0 tail 12 room 4
11:34:54.475 ^Jload:0x40080400,len:6352
11:34:54.478 ^Jentry 0x400806b8
11:34:54.478 ^JRTK Variant: SFE_Surveyor^J
11:34:54.922 Reset reason: ESP_RST_INT_WDT
11:34:54.925 ^JSD init failed. Trying again 1 out of 2^J
11:34:55.431 SD init failed. Trying again 2 out of 2^J
11:34:55.798 SD init failed. Is card present? Formatted?
11:34:55.801 ^JConfig file read failed: SD offline
11:34:55.804 ^JMAX17048 configuration complete
11:34:55.859 ^JBatt (100%): Voltage: 4.19V Discharging: -0.62%/hr Green^J
11:34:59.167 GNSS configuration complete
11:34:59.167 ^JWiFi Stopped
11:35:02.836 ^J[I][BluetoothSerial.cpp:572] _init_bt(): device name set
11:35:03.346 ^JBluetooth broadcasting as: Surveyor Rover-0CE2
11:35:03.352 ^JState: Rover - No Fix
11:35:03.354 ^J[I][BluetoothSerial.cpp:244] esp_spp_cb(): ESP_SPP_INIT_EVT
11:35:03.357 ^J[I][BluetoothSerial.cpp:247] esp_spp_cb(): ESP_SPP_INIT_EVT: slave: start
11:35:03.366 ^J[I][BluetoothSerial.cpp:348] esp_spp_cb(): ESP_SPP_START_EVT
11:35:03.371 ^JBatt (100%): Voltage: 4.19V Discharging: -0.62%/hr Green^J
11:35:03.377 System time set to: May 31 2021 09:35:05
11:35:03.381 ^JState: Rover - Fix
11:35:03.381 ^JRover Accuracy (m): 0.0311
11:35:03.888 ^JState: Rover - RTK Fix
11:35:03.891 ^J[I][BluetoothSerial.cpp:255] esp_spp_cb(): ESP_SPP_SRV_OPEN_EVT: 0
11:35:04.188 ^JClient Connected
11:35:04.188 ^JRover Accuracy (m): 0.0290
11:35:05.908 ^JRover Accuracy (m): 0.0323
11:35:07.924 ^JBatt (100%): Voltage: 4.19V Discharging: -0.62%/hr Green^J
```

Second crash 11:52:11
```
11:51:43.358 Rover Accuracy (m): 0.0220
11:51:43.358 [E][BluetoothSerial.cpp:317] esp_spp_cb(): RX Full! Discarding 334 bytes
11:51:43.802 [E][BluetoothSerial.cpp:317] esp_spp_cb(): RX Full! Discarding 990 bytes
11:51:43.807 [E][BluetoothSerial.cpp:317] esp_spp_cb(): RX Full! Discarding 764 bytes
11:51:43.814 [E][BluetoothSerial.cpp:317] esp_spp_cb(): RX Full! Discarding 342 bytes
11:51:43.881 [E][BluetoothSerial.cpp:317] esp_spp_cb(): RX Full! Discarding 540 bytes
11:51:43.936 [E][BluetoothSerial.cpp:317] esp_spp_cb(): RX Full! Discarding 840 bytes
[... 80 lines cut ...]
12:06:43.636 getPortSettings failed
12:06:43.636 getPortSettings failed
12:06:44.460 Module failed initial config.
12:06:46.083 Module failed to save.
12:06:46.392 Failed to configure module. Trying again.
12:06:46.397 getPortSettings failed
12:06:47.965 GNSS configuration complete
12:06:49.339 WiFi Stopped
12:06:52.314 [I][BluetoothSerial.cpp:572] _init_bt(): device name set
12:06:52.808 Bluetooth broadcasting as: Surveyor Rover-0CE2
12:06:52.814 State: Rover - No Fix
12:06:52.816 [I][BluetoothSerial.cpp:244] esp_spp_cb(): ESP_SPP_INIT_EVT
12:06:52.819 [I][BluetoothSerial.cpp:247] esp_spp_cb(): ESP_SPP_INIT_EVT: slave: start
12:06:52.827 [I][BluetoothSerial.cpp:348] esp_spp_cb(): ESP_SPP_START_EVT
12:06:52.833 Batt (100%): Voltage: 4.20V Charging: 0.42%/hr Green
12:06:52.836 System time set to: May 31 2021 10:06:53
12:06:52.842 State: Rover - Fix
12:06:52.842 Rover Accuracy (m): 0.0648
12:06:53.342 State: Rover - RTK Fix
12:06:53.345 [I][BluetoothSerial.cpp:255] esp_spp_cb(): ESP_SPP_SRV_OPEN_EVT: 0
12:06:54.932 Client Connected
12:06:54.932 Rover Accuracy (m): 0.0364
12:06:55.385 Rover Accuracy (m): 0.0404
12:06:57.416 Batt (100%): Voltage: 4.20V Charging: 0.42%/hr Green
```

nseidle, 2021-06-01:
Thank you very much for this testing and reporting! The level of detail is great. 

tldr; I cannot replicate this issue. I suspect you are using an older RC.

I am trying to replicate your [last post](https://github.com/sparkfun/SparkFun_RTK_Surveyor/issues/26#issuecomment-851485765).

A few questions:

1. Lefebure waits indefinitely for a GGA message before connecting to an NTRIP caster. You report that NMEA is turned off so how are you getting past the initial GGA check in Lefebure? Are you seeing something else in Lefebure? I had to enable the GGA message.
2. Are you sure you're using the RC-May28 firmware? We fixed the [dropped NTRIP bytes](https://github.com/sparkfun/SparkFun_RTK_Surveyor/issues/23) issue and I no longer see `RX Full! Discarding xxx bytes` errors (as shown below):
[image]
I will add a date to the firmware version number on the main menu so that we can be sure in the future.
3. Thank you for being specific with the SD card removed. I had an initial suspicion that the [file size check](https://github.com/sparkfun/SparkFun_RTK_Surveyor/blob/release_candidate/Firmware/RTK_Surveyor/RTK_Surveyor.ino#L393) didn't have a file system semaphore check, and that may be causing the crash, but since the microSD card is not present, this cannot be the issue.
4. I thought maybe you had various messages turned on, or your measurement rate set very low (very high fix rate) that would cause the Bluetooth connection to become overrun but because you included your settings file, I see that's not the case. I'm going to make inclusion of the settings file a future requirement!
5. Most of my testing has been with a stationary antenna and you are moving. I don't think this will effect much but we should consider it.

I'll update the release with a [few changes](https://github.com/sparkfun/SparkFun_RTK_Surveyor/commits/release_candidate). Please [give the latest release](https://github.com/sparkfun/SparkFun_RTK_Surveyor/releases/tag/v1.3) a try.

pyrog, 2021-06-01:
> I suspect you are using an older RC.

Unfortunately not, see Your Workbench in https://github.com/sparkfun/SparkFun_RTK_Surveyor/issues/26#issue-907185137.

> Are you sure you're using the RC-May28 firmware?

Yes, see above.

> Lefebure waits indefinitely for a GGA message before connecting to an NTRIP caster.

oh, your right 👍
I thought that Lefebure have the same behaviour as SW Maps 😕
Thank you to noticed me that issue 😃 

Waiting for CGA frame is an optional behaviour of NTRIP casters.
The SW Maps NTRIP client could optionally send once or periodically  a NMEA CGA message, but I don't use this feature for my tests.

For Lefebure NTRIP Client, I must select the _"Use Android GPS"_ option 🤪

[image] [image] [image]

> You report that NMEA is turned off so how are you getting past the initial GGA check in Lefebure?

As NMEA is off, the only "issue" is that Lefebure couldn't display any values in the 4 info boxes.
[image]

> Are you seeing something else in Lefebure? I had to enable the GGA message.

No yet 😃 

- Lefebure decode only NMEA messages, a feature request could be decode UBX protocol ?
- SW Maps is very unstable:
Yesterday I use it with an USB connexion. It decode UBX messages, but after a while, it stall 😢 

> I will add a date to the firmware version number on the main menu so that we can be sure in the future.

That was the case, but it display only "v1.3", not the day of the release candidate.
[image]

> Most of my testing has been with a stationary antenna and you are moving. I don't think this will effect much

Me too.
But I remembered that there is a dynamic platform model configuration.
You set it to [DYN_MODEL_STATIONARY](https://github.com/sparkfun/SparkFun_RTK_Surveyor/blob/613c84276a596a0068502d3ba7729610841dcfa2/Firmware/RTK_Surveyor/Base.ino#L33) for the base mode, and to [DYN_MODEL_PORTABLE](https://github.com/sparkfun/SparkFun_RTK_Surveyor/blob/613c84276a596a0068502d3ba7729610841dcfa2/Firmware/RTK_Surveyor/Rover.ino#L15) for the rover mode.

But depending of the usage (hiking, riding…) it could be better set to DYN_MODEL_PEDESTRIAN, DYN_MODEL_AUTOMOTIVE, DYN_MODEL_BIKE…? See #29

And maybe the behaviour of the F9P is really different with an (un)appropriate platform setting??

pyrog, 2021-06-01:
> I'll update the release with a few changes. Please give the latest release a try.

Thank you 😎 

[image]

pyrog, 2021-06-01:
First test of [SFE_Surveyor_Settings.txt](https://github.com/sparkfun/SparkFun_RTK_Surveyor/files/6575412/SFE_Surveyor_Settings.txt):

- microSD card **inserted**
- Lefebure NTRIP client connected over **BlueTooth**.
Use Android GPS to send CGA message
- Serial terminal connected over **USB**
- NMEA messages disabled
- UBX messages enabled (nav_dop, nav_hpposllh, nav_pvt, nav_sat, nav_sig, nav_timegps, rxm_sfrbx)

During the trip, both apps are displayed side by side to check if an unusual event happened:
no BT disconnection :sunglasses: 

[image]

Both Lefebure GPS and serial log files are loaded into U-CENTER:
no unknown messages are displayed :smiley: 

[image]

pyrog, 2021-06-01:
First test of v1.3-May 31 2021 with [SFE_Surveyor_Settings.txt](https://github.com/sparkfun/SparkFun_RTK_Surveyor/files/6576518/SFE_Surveyor_Settings.txt):

- microSD card **inserted**
- Lefebure NTRIP client connected over **BlueTooth**.
Use Android GPS to send CGA message
- NMEA messages **enabled** (nmea_gga, nmea_gsa, nmea_gst, nmea_gsv, nmea_rmc)
- UBX messages enabled (nav_dop, nav_hpposllh, nav_pvt, nav_sat, nav_sig, nav_timegps, rxm_sfrbx)

The .ubx logfile contain NMEA and UBX messages "properly" interlaced :smiley: 

U-CENTER don't display any unknown messages, only UBX and NMEA as expected:

[image]

nseidle, 2021-06-01:
Very good! It sounds like everything is working.

Ok. I suspect I did something wrong when creating the v13RC-May28 binaries. Everything seems to be working as expected. Please continue to report any bluetooth issues.

closed 2021-06-01 by nseidle