sparkfun/SparkFun_RTK_Firmware #27: 100 cm led is off in RTK

pyrog, 2021-05-31:
### Subject of the issue
100 cm led is off.

### Your workbench
* v13RC May 28.

### Actual behavior
Note: This led is on during the autotest during startup.

[image] [image]

pyrog, 2021-05-31:
I opened the case and leave a cable between the led and the light pipe.
[image]

closed 2021-05-31 by pyrog

nseidle, 2021-06-01:
Thank you for documenting! Yes, we see this during production as well. If the cables are not routed correctly they can sometimes fall between the LED on the PCB and the lightpipe causing the outside lights to look dim or off. Moving the internal cables should fix the issue.