sparkfun/SparkFun_RTK_Firmware #28: Log filename with .ubx extension

pyrog, 2021-05-31:
### Subject of the issue
I use de default messages logging settings. The file contain only NMEA messages.
File name is i.e. [SFE_Surveyor_210529_171637.ubx.](https://github.com/sparkfun/SparkFun_RTK_Surveyor/files/6568655/SFE_Surveyor_210529_171637.ubx.txt)

### Your workbench
*  v13RC May 28.
*  [Default settings](https://github.com/sparkfun/SparkFun_RTK_Surveyor/files/6568277/SFE_Surveyor_Settings.txt) (log only NMEA messages).

nseidle, 2021-05-31:
Yes, all that seems correct. That is the issue? Do you expect the log file to have a txt extension?

pyrog, 2021-06-01:
> Do you expect the log file to have a txt extension?

This more a feature request than an issue 😄 

> Do you expect the log file to have a txt extension?

Yes or a `.nmea` extension if the file contain only text format.
If the log file contain only a u-blox binary format it could have `.ubx` extension.
If the file contain both formats, it could have i.e. a `.log` extension.

nseidle, 2021-06-01:
Ah, I understand. I am hesitant to implement this. 

* What do you do when a UBX log is already open, and the user turns off all messages but NMEA? Rename the file - but there is already UBX data in the file? Leave in place and name the next log at power up a .nmea file?
* For beginning users, they may see a mixture of file types (nmea, ubx, log, etc) and become confused. I feel like a single file type is preferred. 
 
I'm leaning towards leaving the ubx file extension in place as this covers the most situations. Is there a software that you use that expects a .nmea or .log file extension?

pyrog, 2021-06-03:
> I feel like a single file type is preferred.

Ok, but use a "untyped" extension like `.log`

> Is there a software that you use that expects a .nmea or .log file extension?

Yes :smiley: 
At least [JOSM](https://wiki.openstreetmap.org/wiki/JOSM), the OpenStreetMap editor need an `.nmea` extension to load and display it as "GPS" track.

For advanced users, could be set in the setting list. See https://github.com/sparkfun/SparkFun_RTK_Surveyor/issues/24#issuecomment-853705351

nseidle, 2022-04-05:
Hi @pyrog - I'm going to close this issue out. I don't think we need to support multiple file extensions at this time. Please re-open or comment if you think we need to consider it in the future.

closed 2022-04-05 by nseidle