sparkfun/SparkFun_RTK_Firmware #29: Add dynamic model setting

pyrog, 2021-06-01:
### Subject of the feature request
Add an option in the settings to set the dynamic model for the rover mode.

See:
- [SparkFun_Ublox_Arduino_Library/keywords.txt](https://github.com/sparkfun/SparkFun_Ublox_Arduino_Library/blob/1e70755453a898d8267ced641500c33d377409b7/keywords.txt#L258-L267)
- 3.1.7.1 Platform settings in [ZED-F9P — Integration manual](https://www.u-blox.com/sites/default/files/ZED-F9P_IntegrationManual_%28UBX-18010802%29.pdf)

nseidle, 2021-06-01:
A fine idea. I'll implement in v1.4.

nseidle, 2021-06-07:
Added to v1.4RC. Please test and let me know what you think.

closed 2021-06-07 by nseidle

pyrog, 2021-06-17:
I have done a quick test with the Automotive model.
My impression was that the device lost less the RTK fix.

Drawing the track seem to show that:
[image] [image]
Left **default** model. Right **automotive** model