sparkfun/SparkFun_RTK_Firmware #31: Mechanical issues

Stefal, 2021-06-02:
### Subject of the issue
I had in hand over 10 RTK Surveyors, and I see some mechanical issues:
- Not enough "glue" on the battery
- wires which hide some leds
- SD card difficult to insert/remove because the slot is too far from that enclosure external sides. I have to use a SD to microSD adapter to insert/remove the SD card
- SD card difficult to insert/remove because the hole is too tight
- on/off switches are too "soft", it's easy to switch the rover/base mode by mistake, or if the surveyor is in a pocket or ...
- And the most important to me: please use a **thru hole SMA connector**. I have 1 Surveyor with the SMA connected ripped off, and I'm pretty that you will see more cases like this if you continue to use a surface mounted connector, it's too fragile.

nseidle, 2021-06-03:
Hi Stefal,

Excellent feedback.

> Not enough "glue" on the battery

Good to know. We'll change it out to a better glue.

> wires which hide some leds

This is taken care of in production when we close the lid. If the user opens the case, we have to rely on them to route the wires correctly. In addition, the new [RTK Express](https://www.sparkfun.com/products/18019) removes LED light pipes.

> SD card difficult to insert/remove because the slot is too far from that enclosure external sides. I have to use a SD to microSD adapter to insert/remove the SD card

Yes, this is the challenge of a 'Geneva' type SD socket. Alternatively we could use a friction fit socket (as seen on [this product](https://www.sparkfun.com/products/14030)) but the SD card will extend outside the enclosure by ~3mm. This may lead to damage to the SD card or socket. I'll do some testing. Would you rather have a friction fit socket?

> SD card difficult to insert/remove because the hole is too tight

See above.

> on/off switches are too "soft", it's easy to switch the rover/base mode by mistake, or if the surveyor is in a pocket or ...

They do tend to flip in a backpack or pocket. We are working on two solutions. 1) A carrying case that protects the switches. 2) The [RTK Express](https://www.sparkfun.com/products/14030) removes the switches.

> And the most important to me: please use a thru hole SMA connector. I have 1 Surveyor with the SMA connected ripped off, and I'm pretty that you will see more cases like this if you continue to use a surface mounted connector, it's too fragile.

This should have been your first point ;) You are the second person to report this problem. It's hard for us to replicate the forces needed to rip the SMA SMD connector from its socket. I will look into an [edge mount SMA connector](https://www.sparkfun.com/products/16481) that should improve the reliability.

Please keep the feedback coming! Your thoughts are very valuable to me.

pyrog, 2021-06-03:
Same issues for me, except the SMA connector still alive :smiley: .

- Battery glue:
Use a hook and look tape?

- SD card socket:
Is it possible to move the socket **1 mm** "outside" the PCB?
This is difficult to remove the card, especially if an USB cable is connected to the "Config ESP32"  connector.
2 times the SD card "fall" inside the box :rofl:

- Switches:
I didn't shutdown the surveyor yet, but once I press the Setup switch when inserting it in the backpack.
Use the same switches as the RTK Express?
<br/>Long press on the Setup button to switch from base to rover mode.
Short press: add a "marker" in the log file?

- Edge mount SMA connector:
Sound good to me :smiley:

Stefal, 2021-06-11:
> Yes, this is the challenge of a 'Geneva' type SD socket. Alternatively we could use a friction fit socket (as seen on [this product](https://www.sparkfun.com/products/14030)) but the SD card will [...]

I'm not a big fan of these SD slot type, but If the slot can't be move a little more outside, perhaps it's a solution.

> > SD card difficult to insert/remove because the hole is too tight
> See above.

Not exactly, the hole in the enclosure was blocking the SD Card. I had to unscrew the enclosure a few to be see the SD card popping up outside the slot.

> They do tend to flip in a backpack or pocket. We are working on two solutions. 1) A carrying case that protects the switches. 2) The [RTK Express](https://www.sparkfun.com/products/14030) removes the switches.

Using sliding switch? There are usually more "hard" to move.

> > And the most important to me: please use a thru hole SMA connector. I have 1 Surveyor with the SMA connected ripped off, and I'm pretty that you will see more cases like this if you continue to use a surface mounted connector, it's too fragile.

> This should have been your first point ;) You are the second person to report this problem. 

It was probably me, I've contacted the support the week before I wrote this issue.

>It's hard for us to replicate the forces needed to rip the SMA SMD connector fromits socket. I will look into an [edge mount SMA connector](https://www.sparkfun.com/products/16481) that should improve the reliability.
Hmmm... I assume it's more difficult to build the RTK Surveyor if you use a thru-hole connector, but I still think it's the best solution. Sometime the sma -> Gnss Antenna is rigid, and put some stress on the sma connector. And If you use an helicoidal antenna directly on the sma connector, you have to be very very careful.

nseidle, 2021-06-11:
Ah, yes, using a helical antenna directly mounted to the SMA makes a lot of sense. And I see how that can mechanically strain the connection. I have a proto coming soon to test all this out. Thanks for the feedback!

Stefal, 2021-06-11:
BTW, and completely off topic. There are some articles from you, translated, in a french electronic magazine.
We have some fun here too with GNSS.... see one of my projects [RTKBase](https://github.com/Stefal/rtkbase), and the open base station network we have now : https://centipede.fr/index.php/view/map/?repository=cent&project=centipede

Stefal, 2021-06-12:
> Ah, yes, using a helical antenna directly mounted to the SMA makes a lot of sense. 

I use an adapter to do this:

[image]

pyrog, 2021-06-14:
> They do tend to flip in a backpack or pocket. We are working on two solutions.
>1. A carrying case that protects the switches.
>2. The RTK Express removes the switches.

1. You could use this [switch](https://www.sparkfun.com/products/11310)? :smile: 
[image]

2. Seem the best solution

On a surveyor pole or on the top of a backpack this is not an issue, but I confirm that inside a pocket or a backpack it is a real issue.

Stefal, 2021-06-26:
Hi!
I use the RTK Surveyor mostly for fun, and tests, but yesterday I had to use it to geolocalize 300 trees for a work. 
At one point I stumbled, without falling. I don't know what happened exactly, I probably touched the helical antenna while I was getting my balance back. Result: RTK Surveyor's Sma connector ripped off. The Surveyor is really VERY fragile. 😞 
I'll contact the support next week.

nseidle, 2021-09-20:
SMA Connector: We have redesigned all RTK products to use a non-SMD SMA antenna that is far more robust.

SD Socket: We have a new microSD socket on the RTK Express Plus (soon to be released) that is better. There is not much we can do without changing the thickness of the end plate.

Switches: We have a new [carrying case](https://www.sparkfun.com/products/18246) that should protect them from accidental toggling during transit.

closed 2021-09-20 by nseidle