sparkfun/SparkFun_RTK_Firmware #36: Bad filename

pyrog, 2021-06-14:
### Subject of the issue
The log written at jun, 4th 14:35 have a wrong name: `SFE_Surveyor_7946210131_1986033.ubx`
It was recorded from 2021-06-04T13:41:53.250Z to 2021-06-04T14:24:57.500Z

### Your workbench
* version v1.3-May 31 2021
* settings.txt see https://github.com/sparkfun/SparkFun_RTK_Firmware/issues/25#issuecomment-854996067
* connected over Bluetooth with Lefebure NTRIP Client that transmitting RTCM 3 back to the device
* powered by USB on ESP32 Config port

### Expected behavior
Filename should be `SFE_Surveyor_210604_134153.ubx`

nseidle, 2021-06-14:
Thanks for reporting. Looks like the device created the log file before it got a proper time lock. @adamgarbo would be disappointed in me ;)

I think I know how to fix this. Thanks for reporting!

adamgarbo, 2021-06-14:
Haha! 

The strange thing is the timestamp shown above looks like there are some weird rollovers happening. 

By the way, you can likely ditch lines 805-806. At quick glance, they allow incorrect timestamps through to the log file timestamp and make the if statement on lines 807-808 redundant. If `getConfirmedTime/getConfirmedDate` are true then `getTimeValid/getDateValid` will always be true.
https://github.com/sparkfun/SparkFun_RTK_Firmware/blob/6a282cb298b1752abcc4ac32c5c3caba5866533f/Firmware/RTK_Surveyor/menuMessages.ino#L805-L808

Cheers,
Adam

nseidle, 2021-06-17:
This was fixed in the latest commit. Getting the confirmed Time/Date, and then creating the log name based on them, now take a few more seconds after first fix to be obtained but it should not affect general use.

closed 2021-06-17 by nseidle

pyrog, 2021-06-17:
Thanks a lot