sparkfun/SparkFun_RTK_Firmware #37: Lost data

pyrog, 2021-06-14:
### Subject of the issue
* Some coordinates are incomplete.
* Some data are missing.

Here a digit is lost in 7.1012131, 4.7285551. The real coordinates are **4**7.1012131, 4.7285551.
[image]

After adding the missing digit, the track is correct:
[image]

Here 4 and 3 measurements have missing data (fix type for 2, coordinates for 5):
[image]  [image]  [image]

Here red dots have missing fix type:
[image]  [image] [image]

### Your workbench
* rtkFirmwareVersion=v1.3-Jun  4 2021
* device connected over Bluetooth with several apps that transmitting RTCM 3 back to the device
* everything being powered by USB
* default settings [SFE_Surveyor_Settings.txt](https://github.com/sparkfun/SparkFun_RTK_Firmware/files/6648283/SFE_Surveyor_Settings.txt) (NMEA only)

pyrog, 2021-06-14:
Sample [bretagne.nmea](https://github.com/sparkfun/SparkFun_RTK_Firmware/files/6648664/bretagne.nmea.txt) loaded in JOSM
- 27 coordinates imported
- 1 malformed sentence
- 6 checksum errors

[image]
[image]

nseidle, 2021-06-14:
This looks like a duplicate of [issue 30](https://github.com/sparkfun/SparkFun_RTK_Firmware/issues/30). Please add notes to that issue when possible.

closed 2021-06-14 by nseidle