sparkfun/SparkFun_RTK_Firmware #39: Firmware upgrade : duplicate menu item

pyrog, 2021-06-16:
### Subject of the issue
The menu display twice the .bin file
```
Menu: Update Firmware Menu
1) Load RTK_Surveyor_Firmware_v14RC_June15.bin
2) Load ._RTK_Surveyor_Firmware_v14RC_June15.bin
x) Exit
```
Selecting item 1 or 2 display:
```
WiFi Stopped
Bluetooth turned off
Loading RTK_Surveyor_Firmware_v14RC_June15.bin

No firmware file found
```

### Your workbench
* SparkFun RTK Surveyor v1.3-Jun  4 2021.
* The μSD card is not freshly formatted
(It contain a lot of files in subdirectories)

nseidle, 2021-06-17:
Hmm, I cannot replicate:

[image]

I think you are a Mac user, yes? Did you copy over an invisible file from the RC zip file? Please turn on all files and see what the contents is.

pyrog, 2021-06-18:
> I think you are a Mac user, yes?

👍 (and Linux too)

> Did you copy over an invisible file from the RC zip file?

No, the unzip file don't contain any invisible file.
```
$ ls -al ~/Downloads/RTK_Surveyor_Firmware_v14RC_June15 
total 6080
drwx------@  4 user  staff      128 16 jui 07:20 .
drwx------@ 44 user  staff     1408 17 jui 16:22 ..
-rw-r--r--@  1 user  staff  1520688 15 jui 20:34 RTK_Surveyor_Firmware_v14RC_June15.bin
-rw-r--r--@  1 user  staff  1586224 15 jui 20:35 RTK_Surveyor_Firmware_v14RC_June15_combined.bin
```

But macOS see the `.bin` files as compressed file.
When I drop it to the SD card, it create a `._` file !
```
$ ls -al /Volumes/Surveyor/
total 52224
drwxrwxrwx@ 1 user  staff    32768 18 jui 08:21 .
drwxr-xr-x  5 root  wheel      160 18 jui 08:19 ..
-rwxrwxrwx  1 user  staff     4096 18 jui 08:21 ._RTK_Surveyor_Firmware_v14RC_June15.bin
-rwxrwxrwx  1 user  staff     4096 16 jui 15:43 ._SFE_Surveyor_210616_070432.ubx
-rwxrwxrwx  1 user  staff     4096 16 jui 11:45 ._SFE_Surveyor_Settings.txt
-rwxrwxrwx@ 1 user  staff  1520688 15 jui 20:34 RTK_Surveyor_Firmware_v14RC_June15.bin
-rwxrwxrwx@ 1 user  staff  4678873 16 jui 07:30 SFE_Surveyor_210616_070432.ubx
-rwxrwxrwx@ 1 user  staff     2691 17 jui 15:37 SFE_Surveyor_Settings.txt
```

Does the device should hide hidden files ??

nseidle, 2021-06-21:
Thanks for reporting. I have a fix in place to only show files with the correct leader "RTK_Surveyor_Firmware_". Should fix your problem.

nseidle, 2021-06-22:
Fixed. Please use binaries from [the release](https://github.com/sparkfun/SparkFun_RTK_Firmware/releases/tag/v1.4).

closed 2021-06-22 by nseidle

pyrog, 2021-06-23:
Works fine for me, the `._RTK_Surveyor_Firmware_v14RC_June22.bin` is hidden 😃 

```
Menu: Update Firmware Menu
1) Load RTK_Surveyor_Firmware_v14RC_June22.bin
```