sparkfun/SparkFun_RTK_Firmware #40: portable dynamic model not displayed on menu

Stefal, 2021-06-22:
### Subject of the issue
If you select "portable" as dynamic model, it is not displayed on the menu:
[image]

### Your workbench
* What version of RTK Surveyor firmware are you running? SparkFun RTK Surveyor v1.4-Jun 15 2021

### Steps to reproduce
Tell us how to reproduce this issue. Please include a copy of your settings.txt file used on the unit.

### Expected behavior
We should see "Set dynamic model: portable"

nseidle, 2021-06-22:
Thanks for reporting. I can confirm.

nseidle, 2021-06-22:
Ah, I knew I was forgetting something when I implemented this. The enum starts at 0 and skips 1. I'll fix shortly.

[image]

nseidle, 2021-06-22:
New binaries are in [the release](https://github.com/sparkfun/SparkFun_RTK_Firmware/releases/tag/v1.4). Please keep the issues coming!

closed 2021-06-22 by nseidle