sparkfun/SparkFun_RTK_Firmware #42: RTK Surveyor: display RTK Float state

pyrog, 2021-06-24:
### Subject of the feature request
Display RTK Float and RTK Fixe state.
This is very useful during a hike under trees 😃 

### Your workbench
* RTK Surveyor

### Suggested behavior
The RTK led could blink in Float and be always on in Fixe.

The blink cycle could be:
- 50% 100% …
-  0% 100% …

### Actual behavior
The led is lit only when the GNSS receiver is in RTK Fixe.

nseidle, 2021-06-25:
The RTK LED is driven directly by the ZED-F9P so unfortunately we cannot change the behavior.

[image]

That said, if you plug a [Qwiic OLED](https://www.sparkfun.com/products/14532) onto your RTK Surveyor, magical things will happen :) RTK Float is indicated by a blinking double-crosshair and RTK Fix is a solid double-crosshair.

closed 2021-06-25 by nseidle

pyrog, 2021-06-27:
> The RTK LED is driven directly by the ZED-F9P so unfortunately we cannot change the behavior.

If you design another version of the PCB, is it possible to use ESP32 IOs pins to drive this led ?