sparkfun/SparkFun_RTK_Firmware #43: Cyclic logging

Eric-FR, 2021-07-01:
I would like an option to start a new log file each n minutes, in order to get not too big files. For instance, if you ask for 15 minutes, each 15 minutes, the current ubx file is closed and a new one is created. Of course, new file should start at round hours i.e. 16:45:00, 17:00:00, 17:15:00... for a better readability. And files should be joined togeteher in a daily subfolder, for a better managemnt, with subfolder naming like 210702 for July 2021, the 2nd.

Right now, there is an option to "Set max logging time" that I don't really see the utility.

Eric-FR, 2021-07-02:
Typical use case for cyclic logging is the use of the Surveyor as a temporary base for a couple of days. We will need to record raw data to calculate the precise position of the base and to do post-processing when one can not connect base and rover.

nseidle, 2021-07-02:
**Set max logging time** is most used for PPP for a fixed base. There is no need to have it log for 40+ hours. I use it to stop logging after 10 hours (default). 

Your request makes sense and is do-able. How does this sound? *Max amount of time per log cycle* would have a default of of 0 minutes (default). If user sets it to 90, a new log would be created every 90 minutes.

Eric-FR, 2021-07-04:
Even for PPP, I would suggest 24 hours records to average daily phenomena like earth tilde or atmospheric variation.

I would more suggest something like: _Cyclic log file duration_ with predefined values that are submultiple of day duration. For instance : 5 mn, 15 mn, 60 mn, 4 hours, 24 hours, infinite (default).

nseidle, 2021-07-22:
> predefined values that are... 5 mn, 15 mn, 60 mn, 4 hours, 24 hours, infinite (default).

This is a sure fire way to have you come back and ask for a 20 minute option ;) We try to leave inputs as flexible as possible for all users. But this is still a fine feature request. I need to push it to v15 but it can be done.

nseidle, 2022-03-14:
I think I have this working:

[image]

Above is three files, created 1 minute apart. 

[image]

This is what the menu item looks like. The default value is 1440 or 24 hours, the same as max logging time. 

I did not implement a subfolder filing system or aligning the time cutoffs.

It only took me the release of 7 other versions but we got there :) Please give the release candidate a try and let me know if you see any problems.

Eric-FR, 2022-03-16:
I will have a look with static recordings on my balcony.

Eric-FR, 2022-03-26:
I made a test during 36 hours with 1 hour max log length. Hourly files are present and data inside are fine.

nseidle, 2022-03-28:
Excellent! Thanks for testing.

closed 2022-03-28 by nseidle