sparkfun/SparkFun_RTK_Firmware #44: Galileo missing

Eric-FR, 2021-07-04:
RTK express with RTK_Surveyor_Firmware_v14RC_June28.

Looking at the ubx files recorded on the SD card, there was no Galileo data inside. A closer investigation with ublox center connected "config ublox" showed that Galileo was enabled but E1 and E5b signals were disabled.

I would strongly suggest to enable full Galileo signals by default. My experience with post-processing (RTKLib) is showing that Galileo availability is decisive for the range of fix.

More surprisingly, with a Surveyor I was loaned to, Galileo was fully activated. I asked the owner if he modified something. Waiting for answer.

Eric-FR, 2021-07-05:
BTW, I discovered that F9P firmware of the RTK Express is in version 1.12, not in the last 1.13.

I tried with u-center to update it but failed:
```

Setup pipes

Firmware update tool launched

 -p STDIO -b 9600:9600:9600 --no-fis 1 -s 1 -t 1 -v 1 "E:\Downloads\UBX_F9_100_HPG_113_ZED_F9P.7e6e899c5597acddf2f5f2f70fdf5fbe.bin"

----------CMD line arguments-----------

Image file:        E:\Downloads\UBX_F9_100_HPG_113_ZED_F9P.7e6e899c5597acddf2f5f2f70fdf5fbe.bin
Flash:             <compiled-in>
Fis:               flash.xml
Port:              STDIO
Baudrates:         9600/9600/9600
Safeboot:          1

Reset:             1
AutoBaud:          0
Verbose:           1
Erase all:         1

Erase only:        0
Training sequence: 1

Chip erase:        0

Merging FIS:       1

Update RAM:        0

Use USB alt:       0

---------------------------------------

  0.0 u-blox Firmware Update Tool version 19.03

  0.0 Updating Firmware 'E:\Downloads\UBX_F9_100_HPG_113_ZED_F9P.7e6e899c5597acddf2f5f2f70fdf5fbe.bin' of receiver over 'STDIO'

  0.0   - Opening and buffering image file

  0.0   - Verifying image

  0.0   - image 0 size 675668

  0.0   - image 1 size 517932

  0.0   - config size 0

  0.0   - Trying to open port STDIO

  0.0   - Setting baudrate to 9600

  0.2   - Sending training sequence

  1.2   - Retry poll

  2.2   - Retry poll

  3.2   - Retry poll

  3.2 ERROR: Version poll failed.

  3.2 Firmware Update FAILED

Firmware Update Utility has unexpectedly terminated

Exit code (2)
```

Eric-FR, 2021-07-05:
SFE_Express_Settings.txt
```
sizeOfSettings=424
rtkIdentifier=20
rtkFirmwareVersion=v1.4-Jun 28 2021
zedFirmwareVersion=FWVER:HPG 1.12

```

[image]

SFE_Surveyor_Settings.txt
```
sizeOfSettings=424
rtkIdentifier=20
rtkFirmwareVersion=v1.4-Jun 28 2021
zedFirmwareVersion=FWVER:HPG 1.13

```

[image]

pyrog, 2021-07-05:
> Looking at the ubx files recorded on the SD card, there was no Galileo data inside

Same issue with NMEA only messages.

[SFE_Express_Settings.txt](https://github.com/sparkfun/SparkFun_RTK_Firmware/files/6765487/SFE_Express_Settings.txt)

[SFE_Express_210630_143503 sample.ubx](https://github.com/sparkfun/SparkFun_RTK_Firmware/files/6765527/SFE_Express_210630_143503.sample.ubx.txt)

Eric-FR, 2021-07-05:
For information, what I did for firmware update on the GPS-RTK2 : 
https://forum.sparkfun.com/viewtopic.php?f=116&t=52098&p=212035

nseidle, 2021-07-06:
I haven't had a chance to dig into the Galileo issue, but there are [firmware upgrade tutorials](https://learn.sparkfun.com/tutorials/how-to-upgrade-firmware-of-a-u-blox-gnss-receiver/all) available. I've upgraded many ZEDs. If I remember correctly, a few (v1.1 and older firmware?) needed me to uncheck the 'Enter Safeboot before update' checkbox.

Eric-FR, 2021-07-06:
I updated to the last version of u-center (21.05). I tired again to update ZED's firmware, boot with and without Safeboot option. It failed. 

```
Setup pipes

Firmware update tool launched

 -p STDIO -b 9600:9600:9600 --no-fis 1 -s 1 -t 1 -v 1 "E:\Downloads\UBX_F9_100_HPG_113_ZED_F9P.7e6e899c5597acddf2f5f2f70fdf5fbe.bin"

----------CMD line arguments-----------

Image file:        E:\Downloads\UBX_F9_100_HPG_113_ZED_F9P.7e6e899c5597acddf2f5f2f70fdf5fbe.bin

Flash:             <compiled-in>

Fis:               flash.xml

Port:              STDIO

Baudrates:         9600/9600/9600

Safeboot:          1

Reset:             1

AutoBaud:          0

Verbose:           1

Erase all:         1

Erase only:        0

Training sequence: 1

Chip erase:        0

Merging FIS:       1

Update RAM:        0

Use USB alt:       0

---------------------------------------

  0.0 u-blox Firmware Update Tool version 21.05

  0.0 Updating Firmware 'E:\Downloads\UBX_F9_100_HPG_113_ZED_F9P.7e6e899c5597acddf2f5f2f70fdf5fbe.bin' of receiver over 'STDIO'

  0.0   - Opening and buffering image file

  0.0   - Verifying image

  0.0   - image 0 size 675668

  0.0   - image 1 size 517932

  0.0   - config size 0

  0.0   - Trying to open port STDIO

  0.0   - Setting baudrate to 9600

  0.2   - Sending training sequence

  1.3   - Retry poll

  2.3   - Retry poll

  3.3   - Retry poll

  3.3 ERROR: Version poll failed.

  3.3 Firmware Update FAILED

Firmware Update Utility has unexpectedly terminated

Exit code (2)
```

Eric-FR, 2021-07-09:
After a new try, I was able to update the firmware. I don't know why. May be, because I switch off and on the Express just before???

I will investigate again the Galileo issue.

pyrog, 2021-07-09:
I updated the F9P of the RTK Express with an old and slow PC under Windows 10.
At 4800 bds, the transfer stoped randomly at 1/3.

With the "Use chip Erase" option checked, the update completed successfully 😎 
[image]

Eric-FR, 2021-07-11:
After 2 days recording car drives, I can confirm that Galileo data are available in ubx files, for post-processing with RTKLib.
I would suggest closing although there are still two strange points:

- Why did the Express came with FW 1.2 for the F9P?
- Why was the FW update not straightforward?

pyrog, 2021-07-11:
>  possibility to update the F9P firmware through the SD card (and the ESP32).

Could be really cool 😎
But I don't think that u-Blox published the protocol??

nseidle, 2021-07-22:
> Why did the Express came with FW 1.2 for the F9P?

There is a world wide shortage of everything. We are getting ZED-F9Ps but some have v1.12 firmware on them. Currently, the difference is not essential to the performance of the RTK line so we don't both with updating the firmware.

> Why was the FW update not straightforward?

Hard to say.

> possibility to update the F9P firmware through the SD card

Ya, this is probably not going to happen anytime soon. We'll leave the firmware updating process to u-center.

nseidle, 2021-07-22:
Now to the original post topic - v1.4 RC now has a sub menu for configuring the constellations. I learned far more than I wanted to know about that particular corner of the ZED-F9P firmware (wow it's not well documented) but it's done.

[image]

closed 2021-07-22 by nseidle