sparkfun/SparkFun_RTK_Firmware #45: Changing mode between base and rover is too easy

Eric-FR, 2021-07-11:
It is very to change from rover to base... inadvertently. You just need a short press on one of the two buttons. That's happened to me yesterday. I missed the switch off by not pressing long enough but I didn't saw that it switched to base mode. So pressed long to turn off. When I started again, I didn't noticed that the Express was in base mode, even more that I was in an underground parking. While I drove outside, I noticed that Lefebure Ntrip client was disturbed with several strange parameters. Finally, when I stopped driving, I was able to investigate and I discovered that the Express was in base mode. After switching back to rover, all was ok.

Considering that switching between base and rover is not a common task, I would suggest, at least, to wait for a long press to switch between base and rover.

nseidle, 2021-07-22:
Good feedback. I'll see what I can do. I'll push this to v15.

nseidle, 2021-07-23:
I think I've got a good fix for this. I've added a graphical menu for setup to allow for additional states (the bubble level for example).

[image]

Changing to Rover or Base now requires some user interactions more than a random bump. Please re-open if this is not adequate.

closed 2021-07-23 by nseidle

pyrog, 2021-09-25:
Tested the Bubble mode with v1.7RC: this is so nice, thank you :smiley

nseidle, 2021-09-25:
You're welcome! It's a bit silly since you need to be sure the device is orthogonal to the stick but it's handy (and just fun).