sparkfun/SparkFun_RTK_Firmware #46: Need a method to mark waypoint

Eric-FR, 2021-07-11:
When using the Express as stand-alone, there is no way to mark a waypoint. Even when using with a smartphone that is only running the NTRIP client, I want to record waypoints quickly, on the fly, just by pushing one button.

I would suggest that a short press on whatever button you want while record a waypoint. Ideally trigger an event on the INT pin of the F9P so one can directly have it in the UBX files (UBX-TIM-TM2 message).

pyrog, 2021-07-11:
Or simply logging an event in the SD card?

nseidle, 2021-07-22:
I am very hesitant to do this. The RTK line is designed to be used in conjunction with a GIS app, not on its own.  Additionally, there is no button on the Surveyor so this feature will only apply to Express

nseidle, 2021-07-23:
Ok, I'm becoming ok with this. It's proving to be a nice solution to issue #45 as well.

[image]

What do you want the event to say? We're limited to around 82 characters.

[image]

In addition, v1.5 there is now a menu that pops up when you press the setup button. Rather than toggling between Rover/Base, the user can now select which mode to enter. 'Mark' is at the top. So if you press the Setup button, the menu opens. If you do nothing for 1.5 seconds, the menu will select 'Mark' and record the event -> Single button recording (and it prevent a user from automatically switching to Base).

Please give [v1.5 RC](https://github.com/sparkfun/SparkFun_RTK_Firmware/releases/tag/v1.5) a try. It's not thoroughly tested but the basics are working.

Eric-FR, 2021-07-23:
Ok. I will try to install the RC tomorrow so I can check it during a one week travel in South of France.

pyrog, 2021-07-30:
> there is no button on the Surveyor so this feature will only apply to Express.

A quick base/rover could trigger a waypoint marking.
In a future hardware release you could replace this 2 states switch by a single push button.

nseidle, 2021-08-06:
Closing. This is implemented. Please open a new issue if you need modifications.

closed 2021-08-06 by nseidle