sparkfun/SparkFun_RTK_Firmware #49: Checksum error in log file

Eric-FR, 2021-07-23:
When trying to convert ubx files logged with Surveyor or Express to .pos file, I often get a checksum error (script : https://github.com/Stefal/ubx-to-gpx).

When logging with the adalogger, I never see such an error (logger : https://github.com/PaulZC/F9P_RAWX_Logger).

Error message:
[image]

Sample file displaying one checksum error:
[SFE_Express_210704_224834.zip](https://github.com/sparkfun/SparkFun_RTK_Firmware/files/6871381/SFE_Express_210704_224834.zip)

Right now, it is not critical as I still get the expected .pos file. RTKLib is also not complaining.

nseidle, 2021-07-29:
I believe this is related to [previous issues](https://github.com/sparkfun/SparkFun_RTK_Firmware/issues/41#issuecomment-869889194). It will be very hard to eliminate all possible logging errors while supporting BT SPP in both directions at unknown throughput rates.

Just so I understand, you saw 1 error in 1MB log file? That's a good data point for me to know.

And you used Stefal's conversion script to convert UBX to POS. Ok, neat. That could be a much easier tool to test checksum errors. 

What are the 'second must be in 0..59' errors?

Eric-FR, 2021-08-06:
Concerning the previous issue with BT congestion, I don't know if this is related. I'm using 1 Hz for all frequencies/parameters so need for bandwidth should be limited. I will check circumstances on some samples.

About the '0..59' error, this seems to be related to underlying libraries used for reading and converting ubx files, something like time was  10 mn 59 sec 999 1/1000s. After conversion and rounding, it should end up like 10 mn 60 s (instead of 11 mm 00 s).

Eric-FR, 2021-08-09:
I made a new try this night with static recording outside of the house and no BT/RTCM correction. I still get checksum errors (20 for 10 hours, acquisition rate at 1 Hz).
[image]

pyrog, 2021-09-26:
During [several test](https://github.com/sparkfun/SparkFun_RTK_Firmware/issues/54#issuecomment-927302653) in v1.7RC I got corrupted data.

```
$GNGST,075122.25,63,5.5,2.4,74,1.1,2.2,2.9*46
” {C  ÇW˚"  P†      Ä Vé†àòåîì‰s.*Ñ/Ó∂{Ã‹øú>ΩpfË0‹ﬂ‡∞Ã†¨|ıÜÛy¸áØÔ‰˛*}˜Ûh\@‡- (;{HLj™™†™†7œDq8∆vƒ–                  "”” &C¿ …Qo‚            h(ÿﬁ±ƒ%ƒ~Wî’XÜÓ#” mD` ÇW˚"  (@ B      v˙≤™zÚ™vﬁâ±	ˇ1)ah¬M•W‘ıQ›#e«>jç‘+·»«É–ﬁ€z˛‘1ˇd;@`ÿˆÉÖ/Î“ü©mUUKAU@E4≤Îè,ı5             ‰€q” F@ ÇW `             Rd#B·¨˙R¿pÙÉ” L‡ ÄÌÌ÷
$GNRMC,075123.00,A,4634.3597352,N,00544.7536378,E,0.030,,250921,,,A,V*1F
$GNGGA,075123.00,4634.3597352,N,00544.7536378,E,1,12,0.68,547.180,M,47.205,M,,*41
```

```
$GNGST,075123.00,63,5.5,2.4,74,1.1,2.1,2.9*43
™é‡;`ÄlìÅÀíÒòIË’UMAU@E4≤Îè,ı5             ‹i” F@ ÇW0              Rd)¡†π“¿N¯n” L‡ ÄÌÌ÷â∞ 1(¸-˜«∞Lï)xﬁÅ≥ÉŒ≤$i¸ÉÓ∫Æﬁ™é‡;`ÄlìÅÀíÒòIË’UMAU@E4≤Îè,ı5             ‹i” F@ ÇW0              Rd)¡†π“¿N¯n” L‡ ÄÌÌ÷
$GNRMC,075124.00,A,4634.3597384,N,00544.7535659,E,0.018,,250921,,,A,V*1C
$GNGGA,075124.00,4634.3597384,N,00544.7535659,E,1,12,0.68,547.267,M,47.205,M,,*42
```

```
$GNGST,075124.00,61,5.4,2.4,74,1.1,2.1,2.9*47
” äC  ÇXb  P†Ä     Ä V£®"&#%$•¯ú£Ü†Î€¥´ûD:Ê%†Cê∆‚}ƒYî˝Ò9“Äx⁄·◊»ÇNRÚH5ñ¿øî}√¿òRÈpú} òe’’–]–0_=2D‚˘[M                         ,¿@” &C¿ …Qè"            h(ÿ÷©˛_≈G˛Nˇò’éZﬂè” mD` ÇXb  (@ B      v˙≤™zÚ™v‚â∞™1(ñÈ-Aº7UÊ˙QﬂˇÏ¿]z’Ù¥@òY∞ˇJzÍÔºH;|ﬂ˚ Ò¸oÍl7UOAW@ET≤Îo¨ı5             ˚¯–” &F@ ÇW?†            iâ>‰Å£®ƒ_£û·î™•ÕX‚ﬂè
$GNRMC,075125.00,A,4634.3597370,N,00544.7535167,E,0.032,,250921,,,A,V*14
$GNGGA,075125.00,4634.3597370,N,00544.7535167,E,1,12,0.68,547.320,M,47.205,M,,*40
```
…

In the car return trip, corrupted data appears when I tried to start WIFI AP (the car was parked at the red dot)
[image]

pyrog, 2021-09-26:
35  errors during the first minute of the [test](https://github.com/sparkfun/SparkFun_RTK_Firmware/issues/54#issuecomment-927354389).

[image]

[SFE_Surveyor_210926_140214.ubx](https://github.com/sparkfun/SparkFun_RTK_Firmware/files/7232057/SFE_Surveyor_210926_140214.start.of.the.log.with.errors.ubx.txt)

nseidle, 2021-10-01:
I have some decent progress to report. I believe I've reduced, and possibly eliminated, the number of write errors. I need to run some very long logs but it looks promising.

Changes I made:
* Increase size of RX buffer from 2k to 6k (with room to increase as needed)
* Move file synchronization and file date update out of the read task and into the main loop (file is updated every 5 seconds).
* Increase default SPI speed to 16MHz. This may cause issues with some SD cards so let me know if you see an issue. The speed can be lowered through the debug menu as needed.

[image]

Eric-FR, 2021-10-01:
But, after upgrade, SPI frequency is still at 8 MHz in debug menu. I will make some testing at this frequency before increasing it to 16 MHz.

nseidle, 2021-10-02:
The default was increased but your SD card settings file has it set to 8MHz. If you want, you can do a factory reset and it will make it 16MHz.

pyrog, 2021-10-04:
Default settings (NMEA at 4 Hz)

Version | Distance | Duration | Errors|Comment 
--|--|--|--|--
v1.7 | 48 km | 45 mn | 0
v1.7 | 38 km | 3 days | 12| 2 records spaced 3 days apart sharing the same file
v1.8RC | 38 km | 35 mn | 0|Without RTCM stream
v1.8RC | 38 km | 40 mn | 0|

The last test in v1.7 with errors was in fact a trip of 48 km and 48 mn, and a record of 87 seconds taken 3 days after (during WIFI AP test ??)

So it seem that since V1.8RC there is no corrupted data 😃

pyrog, 2021-10-05:
Version | Distance | Duration | Errors|Comment 
--|--|--|--|--
v1.8RC | 34 km | 31 mn|2|No binary data, only NMEA checksum errors

pyrog, 2021-10-06:
Version | Distance | Duration | Errors|Comment 
--|--|--|--|--
v1.8RC | 7 km | 1h47|0|Hiking
v1.8RC | 41 km | 1h14|0|Car trip

Eric-FR, 2021-10-07:
Last Monday, I made a car travel 2x300 km. I didn't found any checksum error in UBX files (and also no reboot).

v1.8RC-Oct1