sparkfun/SparkFun_RTK_Firmware #50: Configuration menu

Eric-FR, 2021-07-29:
With the Express and the Firmware_v15RC-July23.bin, we have access to the menu through the Setup button. When going down, there is a Configuration entry. When selecting it, the screen is just switching off but the receiver is still running. If pressing Setup again, the screen switch on and menu appears again, to mark a waypoint for instance, but then switch off again. By using Base or Rover options, the screen switch on and stay on.

So, I don't know for is the configuration menu, but for sure, it don't behave as expected.

nseidle, 2021-07-29:
Indeed. It's currently WIP.

nseidle, 2021-08-06:
Closing - the [WiFi AP](https://github.com/sparkfun/SparkFun_RTK_Firmware/issues/51) is now available.

closed 2021-08-06 by nseidle