sparkfun/SparkFun_RTK_Firmware #52: Error: Display — Can't shutdown the RTK Express

pyrog, 2021-08-11:
### Subject of the issue
I try the latest [v15RC firmware](https://github.com/sparkfun/SparkFun_RTK_Firmware/releases/tag/v1.5) and the WIFI connection from the AP.

### Steps to reproduce
1. Connect to the AP from my smartphone
2. Remove the SD card (I don't think that is an issue)
3. Shutdown the device with the Power button : do nothing
4. Press the Exit to rover mode from the config webpage
5. Press again the Power button

The device display:
```
Error:
Display
```
### Actual behavior
I couldn't stop it.
The wifi doesn't respond.
The serial terminal doesn't respond too 😢 

The USB terminal display after a long press on the Setup or Power buttons:
```
State: Shut Down
[…]
State: Shut Down
State: Shut Down
State: Shut Down
State: Shut Down
ButtonCheckTask unknown system state: 26

State: Rover - Not Started
```

nseidle, 2021-08-12:
Thank you for reporting! Sorry you had to open the unit. In the future, you can press and hold the power button for ~10 seconds, then release. The unit will shut off. The power circuit was designed to be independent of the ESP32 so if the firmware fails, the power circuit can still be shut off (you just have to be patient :)

You have good error reporting. State 26 is STATE_SHUTDOWN. You got in a sub state of the [setup state machine](https://github.com/sparkfun/SparkFun_RTK_Firmware/blob/release_candidate/Firmware/RTK_Surveyor/Tasks.ino#L263). Then you hit a hard stop [here](https://github.com/sparkfun/SparkFun_RTK_Firmware/blob/release_candidate/Firmware/RTK_Surveyor/Display.ino#L95) because I don't catch shutdown as a state. There's a few things to fix there but all are easy. Please give me a few hours.

pyrog, 2021-08-13:
> you can press and hold the power button for ~10 seconds, then release
I did that, but probably not enough longer 😅 

One disturbing thing is that a long press on `Setup` button will shutdown the device (same behaviour as pressing `Power` button during a long time).

According to #51 it should enter the config mode.

pyrog, 2021-08-13:
I tried to reproduce this issue without success. The differences:
- I use Safari on my laptop
- There is not SD card
- I don't use the serial terminal

The `Power` and `Setup` button have the correct behaviour when the computer is connected to the WIFI AP 😄 
I founded a minor issue: #53

nseidle, 2021-08-13:
Ok, I think I've got this fixed in RCAug13. I was able to reproduce a few scenarios where the device would hang with 'Error: Display'. I also managed to get the unit into an infinite Mark_Event loop which was fun. All sound be good. 

> One disturbing thing is that a long press on Setup button will shutdown the device (same behaviour as pressing Power button during a long time).

Fun fact: either setup or power will do setup, and either setup or power buttons will power down the unit. It was left over from some testing I was doing. It's been removed from this RC (Setup now only does setup, Power only does power down).

Please continue to put the device in every configuration you can think of and report issues!

closed 2021-08-13 by nseidle