sparkfun/SparkFun_RTK_Firmware #53: Blank config web page

pyrog, 2021-08-13:
### Subject of the issue
When connecting to 192.168.1.1 the when browser display a empty web page.

### Your workbench
* 1.5RC

### Steps to reproduce
1. Connect to the 192.168.1.1
2. Press the `Setup` button to leave the AP
3. Connect to the 192.168.1.1
4. A blank page is displayed

### Actual behavior
In fact the computer automatically switches to the local WIFI AP.
The 192.168.1.1 corresponds to the manager of the internet box which redirects to the http://192.168.1.1/homeAuthentication.html page.

The issue is in fact that the RTK Express don't display a `404 page` or don't redirect to the config page.
The device return an `500 Internal Server Error`

pyrog, 2021-08-13:
I retry with the SD card inserted after bootstrap.

http://192.168.1.1/test return an error 404 page 🤔

nseidle, 2021-08-13:
There is no a way to have the RTK Express to display a page because it is no longer on any WiFi network after step 2.

1. Connect to the 192.168.1.1
2. Press the Setup button to leave the AP
3. Connect to the 192.168.1.1
4. A blank page is displayed

Once you press Setup to leave the AP (step 2), your phone will attempt to connect to any known WiFi AP (your home WiFi, etc). Once connected to a different AP, if you type 192.168.1.1 into your browser, any number of things will happen because you are no longer connected to RTK Config. You're telling your browser to go poke at 192.168.1.1. There's often a router at that AP. 

Now if you think users would prefer it, we can have any IP: 192.168.4.1 (is the ESP32 standard) or 192.168.1.8, it's trivial. I was trying to make it easy (repeated 1s). I'd like to avoid double digits (because I'm out of pixel width! 😂) but it's possible.

Eric-FR, 2021-08-20:
Hopefully, my local network is on 192.168.0.x so I didn't get confused.

But, I support the idea to go to 192.168.4.x if it is less used.

nseidle, 2021-09-20:
IP is now 192.164.4.1.

closed 2021-09-20 by nseidle