sparkfun/SparkFun_RTK_Firmware #57: Display icon according to dynamic

Eric-FR, 2021-09-18:
In rover mode, display an icon corresponding to the dynamic mode selected instead of the present car

Some ideas:

1) Portable : gnss antenna?
2) Stationary : a tripod
3) Pedestrian : a walking guy
4) Automotive : a car (the present icon)
5) Sea : a boat
6) Airborne 1g : an horizontal plane
7) Airborne 2g : a plane going up at 45°
8) Airborne 4g : a plane going up verticaly
9) Wrist : a watch?
10) Bike : a bicycle

;-)

nseidle, 2021-09-19:
I really like this idea. My only concern is the default (portable = GNSS antenna) may be too alien to most users. 

Hmm, it's growing on me. I'll see what I can do.

Eric-FR, 2021-09-19:
Portable (default) : Applications with low acceleration, e.g. portable devices. Suitable for most situations.

Icon : a device/smartphone handle in a hand? a gnns antenna handle by a hand?

For sea, add waves under the boat. It is supposed to be at 0 m level, in the sea, not in a lake.

pyrog, 2021-09-22:
> For sea, add waves under the boat. It is supposed to be at 0 m level, in the sea, not in a lake.

Really?
Maybe just elevation is ~ constant?

Eric-FR, 2021-09-22:
At sea: Recommended for applications at sea, with zero vertical velocity. Zero vertical velocity assumed. Sea level assumed.

nseidle, 2021-10-01:
Ok gents, how good is your pixel art? I need your help coming up with some rough bitmaps for each of these.  Size requirements: Must be equal to or smaller than 15 pixels wide, 12 pixels tall. I've attached my [DynamicModel-Stationary-Example.zip](https://github.com/sparkfun/SparkFun_RTK_Firmware/files/7268559/DynamicModel-Stationary-Example.zip) as an example.

nseidle, 2021-10-01:
These have been implemented in v18RC so you can see what they look like.

nseidle, 2021-10-29:
If anyone has better graphics, please let me know!

closed 2021-10-29 by nseidle