sparkfun/SparkFun_RTK_Firmware #60: RTK Surveyor: How to enter the WIFI Access Point?

pyrog, 2021-09-25:
### Subject of the issue
This is difficult to start the WIFI AP.
Toggle too fast the Setup button do nothing, too slow start the base mode.

### Your workbench
* [v1.6](https://github.com/sparkfun/SparkFun_RTK_Firmware/releases/tag/v1.6)
* [v1.7RC](https://github.com/sparkfun/SparkFun_RTK_Firmware/releases/tag/v1.7RC)

### Steps to reproduce
In V1.6 I toggled quickly the setup button to enter the WIFI AP.
- I could display the beginning of the web page in my web browser, but the connexion was interrupted
- Second attempt work fine. I updated to v1.7RC
[image] [image]

### Expected behavior
Find a way to clearly select WIFI AP or base mode with the Setup switch.

E.g. leave the setup button in 1 state (base) and enter the AP. If there is no WIFI connexion after i.e 30 s, enter the base mode.

Or in Setup position 1 activate the WIFI AP and Setup position 0 use one preset in a settings list (see my comment in [#24](https://github.com/sparkfun/SparkFun_RTK_Firmware/issues/24#issuecomment-853705351)).
Each settings could be a file with a `.SFE_Settings.txt` extension (or starting with the `SFE_` prefix)

### Actual behavior
- Power on the RTK Surveyor: the BT LED blink "quickly"
- Toggle quickly the Setup button: the BT LED fade in/out "slowly" (see [Tasks.ino#L149](https://github.com/sparkfun/SparkFun_RTK_Firmware/blob/b98dc49332539ec8f929d956ff76fcb02b419548/Firmware/RTK_Surveyor/Tasks.ino#L149))

pyrog, 2021-09-30:
After a deeper test, the current behaviour after a quick toggle of the Setup button:
- immediately turned off the Pair LED and display the following messages on the USB console:
```
State: WiFi Config Not Started
Bluetooth turned off
```

- After ~10 seconds, start fade in and out the BT LED and display the following messages on the console:
```
AP Started with IP: 192.168.4.1
State: WiFi Config
```

- At ~15 seconds (from the toggle), my phone or my laptop display the WIFI network "RTK Config"

### Expected behavior
Start fade in BT LED immediately (or e.g. blink another LED) to show that the WIFI AP is started

nseidle, 2021-10-01:
Good points. I had set the LED to pulse when the AP had finished setup and was ready for connections. I can certainly move the fade so it starts as AP starts config. But as you've experienced, the user might try to connect before the AP is ready for connections. 

I think the AP start is fast enough that a user won't notice. I'll give it a try.

nseidle, 2021-10-01:
Oh, and as a side note: the time it takes to enter AP mode is largely dictated by how long it takes to run the freeClusterCount() so that we can show the user how much free space there is on the SD card on the AP HTML page. This freeClusterCount() can take a *long* time, like ~10 seconds. So you should see your unit enter AP mode very quickly when the SD card is not inserted into the device.

pyrog, 2021-10-02:
> So you should see your unit enter AP mode very quickly when the SD card is not inserted into the device.

It took around 3 seconds to enter AP mode without SD card (that's better 😀)
Personally, I always logged data to the SD card.
(Also inserting and removing car is not obvious, and sometime the card fall inside the cas 😢).

So I suggest to display the HTML page "immediately", and update it with the freeClusterCount() later.

> I can certainly move the fade so it starts as AP starts config.

Yes please.

> the user might try to connect before the AP is ready for connections.

This is not really an issue.

The issue is that the user don't really know if the toggle was fast (or slow) enough to enter the AP.

nseidle, 2021-10-29:
This was implemented in v1.8. Please let me know if the user experience is still a problem.

closed 2021-10-29 by nseidle