sparkfun/SparkFun_RTK_Firmware #61: WIFI AP Captive Portal

pyrog, 2021-09-25:
### Subject of the issue
When connecting a device (computer, smartphone) to a RTK Surveyor/Express, display the setup web page automatically.

This is related to #53.

### How to code this?
See [ESP: Auto login/accept message by OS with redirect to page like public WIFI portals access points](https://stackoverflow.com/questions/54583818/esp-auto-login-accept-message-by-os-with-redirect-to-page-like-public-wifi-port) 😃

nseidle, 2021-09-25:
I originally implemented a captive portal but it was not a good experience. I felt like it depended on how the phone handled CPs and that could vary between devices. If I get time I will implement a basic demo and have you try it out.

pyrog, 2021-09-26:
Captive portal may be are not the best solution?
But using a redirection on a standard web page will be enough: it avoid to type and know the exact URL or IP.

So it will be interesting to test this on various devices 😃