sparkfun/SparkFun_RTK_Firmware #62: Power saving mode

pyrog, 2021-09-25:
### Subject of the issue
To reduce power consumption, set RTK Surveyor LEDs or RTK Express display "**off**" after a timeout.
Also in the night, RTK Surveyor LEDs are too bright.

### Expected behavior
Add parameters to reduce brightness of LEDs/Display after i.e. 30s, 1 mn… or never.

If a key is pressed (or the IMU detect a movement in base mode?), the brightness is reset to 100% and the timer is restarted.

So this need 1, 2 or 3 parameters:
- Max and/or min Brightness
- A timeout

Timeout could be:
- user settable (0 never, 1…n seconds) or
- selectable in a list (never, 10 s, 30 s,1 mn)

Possible behaviours:
- Brightness is 100% and fall to min value after the timeout
- Brightness is 100% and fall to 0% after the timeout
- Brightness is max value and fall to min value after the timeout
- Brightness is max value and fall to 0% after the timeout
[image]

nseidle, 2021-10-01:
I really like your mockup! Thanks for that. As a side topic, if you're savvy with HTML and CSS please feel free to 'clean up', modify, or propose changes to the AP config page and do a PR.

WRT LED brightness - Is this really an issue? How many users are going to be surveying or logging at night? 

I cannot change the brightness of the RTK Surveyor LEDs via software. Some are software controlled, others are not (charge LED, PPS, RTK).

WRT to power saving - The total consumption of power from the LEDs or OLED is 1/50th of the total power of the system (the ZED and ESP are power hogs, LEDs and OLED not so much).

I think the user experience would be worse if a user saw the display 'turn off' unexpectedly. If I saw that, I would fear that the unit had powered off or malfunctioned in some day. I'm going to say no to this feature request for now but if you or other users would like to offer other ideas or use cases that I'm not aware of, please do so.

closed 2021-10-01 by nseidle

pyrog, 2021-10-02:
> WRT device brightness - Is this really an issue? How many users are going to be surveying or logging at night?

Eric is a caver (I was). So we could survey thing during the night (especially in winter) after our work :she