sparkfun/SparkFun_RTK_Firmware #64: E (73567) wifi_init: Failed to deinit Wi-Fi driver (0x3001)

pyrog, 2021-09-30:
### Subject of the issue
Unappropriated error message display on the USB console before and after updating the firmware.
Related to https://github.com/espressif/esp-idf/issues/5609.

### Your workbench
* [v1.7RC](https://github.com/sparkfun/SparkFun_RTK_Firmware/releases/tag/v1.7RC).
* RTK Express (probably RTK Surveyor but I can't test it)

### Steps to reproduce
```
SparkFun RTK Express v1.7-Sep 23 2021
** Bluetooth broadcasting as: Express Rover-E536 **
Menu: Main Menu
1) Configure GNSS Receiver
2) Configure GNSS Messages
3) Configure Base
4) Configure Ports
5) Configure Logging
6) Display microSD contents
d) Configure Debug
r) Reset all settings to default
f) Firmware upgrade
x) Exit

Menu: Update Firmware Menu
1) Load RTK_Surveyor_Firmware_v17.bin

x) Exit
1
```

```
E (73567) wifi_init: Failed to deinit Wi-Fi driver (0x3001)
WiFi Stopped
Bluetooth turned off
```

```
Loading RTK_Surveyor_Firmware_v17.bin
…
SparkFun RTK Express v1.7-Sep 27 2021
```

```
E (3340) wifi_init: Failed to deinit Wi-Fi driver (0x3001)
WiFi Stopped
Bluetooth broadcasting as: Express Rover-E536
```

### Expected behavior
As the WIFI wasn't used, this error message shouldn't be displayed

nseidle, 2021-10-01:
Fixed in next v1.8RC.

closed 2021-10-01 by nseidle

pyrog, 2021-10-01:
Thank you :smiley: