sparkfun/SparkFun_RTK_Firmware #65: Crash/reboot during configuration

Eric-FR, 2021-10-03:
With Express, just after upgrade to V1.8RC-Oct1 using USB/serial, I check the receiver configuration through serial. When selecting the debug menu, I got a freeze of serial interface. Later, I was able to connect again through serial and go inside the debug menu.

Next day, I tried to configure through wifi. I also got several crashes with reboot with reboot counter increasing on the display. Crashes are not reproducible. They can append at every moment like webpage loading, parameters saving or back to rover mode.

(BTW, I was not able to check all icons for dynamic due to numerous crashes).

pyrog, 2021-10-04:
I also got something similar issue with the RTK Surveyor after updating v1.7RC to v1.7.

First I tried from WIFI AP (the file was copied at home on the SD card).
The red Base LED stay on.

I retried at home from USB but the terminal stop responding.
I retried later with success.
I thought that it was a USB cable issue.

pyrog, 2021-10-04:
Quick test this morning with the RTK Surveyor
- USB update from v1.7 to[ v1.8](https://github.com/sparkfun/SparkFun_RTK_Firmware/releases/tag/v1.8RC)
- WIFI update back to v1.7
- WIFI update from v1.7 to v1.8

No obvious crash detected 😃 

Quicker test with the RTK Express
- WIFI update from v1.7 to v1.8

Found this minor issue #66

pyrog, 2021-10-06:
Personally I used `esptool` only once (python code from Linux or macOS).
I copy file on SD and use the update firmware menu over USB (and for few tests, use the menu over the WIFI AP)

nseidle, 2021-10-29:
Ah, ok so the unit was resetting after multiple configs. Yep, I can replicate. Let me see...

nseidle, 2021-12-13:
I've been working on this off and on. The immediate solution is 'don't configure the device over WiFi twice during one boot' but I get that you might want to access it multiple times. I believe it's related to how we release the resources for the WiFi and asynchronous web server. For now, I am going to close with a 'won't fix' tag so that we can focus on other features. If you continue to see unexpected or bad behavior, please open a new ticket.

closed 2021-12-13 by nseidle

Eric-FR, 2021-12-14:
OK. I was configuring the device twice just to check the shape of icons for dynamic. I will be careful not to use Wifi config several times at once.