sparkfun/SparkFun_RTK_Firmware #66: SSID message still displayed after firmware upgrade over WIFI

pyrog, 2021-10-04:
### Subject of the issue
After upgrading a RTK Express from v1.7 to [v1.8RC](https://github.com/sparkfun/SparkFun_RTK_Firmware/releases/tag/v1.8RC) the following message is still displayed but the WIFI AP is off (and the WIFI icon is not displayed).
```
   SSID:
RTK config
   IP:
192.168.4.1
```

If I press the Setup button to i.e. select Mark, the SSID message appear again.
My laptop doesn't see this AP.

### Expected behavior
Leave the AP mode.

### Note:
When entering the AP mode, to leave it choose one of the following :
- Press the "Exit to Rover Mode" from the web page
- Press the Setup button and select Rover
- Stop and restart the RTK Express

nseidle, 2021-10-05:
I don't quite understand. Are you saying "After upgrading" via the AP 'Update Firmware' Button, the device stops responding?

You're "Note: ... to leave it choose one of the following" makes it sound like the 'Select Rover' method allows you to exit AP mode.

I guess I need to know, how can I replicate this issue?

nseidle, 2021-10-05:
Steps taken:
* Load a unit with v1.7 via esptool
* Load v1.8RC over WiFi AP config page
* Update device to v1.8RC over WiFi AP config page

Device restarted normally. Hmm, but if the firmware failed to load or doesn't pass CRC check, it would error out... Ah yes, the unit gets into a weird state if the firmware load fails. Fix in the works!

nseidle, 2021-10-05:
In next RC. I am not able to fully test it because I can't replicate it so please re-open if you see this issue again.

closed 2021-10-05 by nseidle