sparkfun/SparkFun_RTK_Firmware #68: Write failed. Binary may be incorrectly aligned

pyrog, 2021-10-13:
### Subject of the issue
Can't update RTK Surveyor from 1.8RC to 1.8

### Steps to reproduce

1. From WIFI AP, try to update the firmware to v1.8
- the Base led stay on
- connect the device to the computer with USB cable
- the following message on the terminal show that the firmware wasn't updated
`SparkFun RTK Surveyor v1.8-Oct  1 2021`
- try to update the firmware from the terminal
```
Menu: Update Firmware Menu
1) Load RTK_Surveyor_Firmware_v18.bin
x) Exit
1
WiFi Stopped
Bluetooth turned off
Loading RTK_Surveyor_Firmware_v18.bin

[141646][D][Updater.cpp:133] begin(): OTA Partition: app0
Moving file to OTA section
Bytes to write: 1830384
[=0%
[==5%
[===10%
[====15%
[=====20%
[======25%
[=======30%
[========35%
[=========40%
[==========45%
[===========50%
[============55%
[=============60%
[==============65%
[===============70%
[================75%
[=================80%
[==================85%
[===================90%
[====================95%
[=====================100%
[======================105%
Write failed. Binary may be incorrectly aligned.

File move complete
Error Occurred. Error #: 2
```

### Note
I updated with success the RTK Express from WIFI AP before.
I used the same SD card (so the same file) to update the RTK Surveyor

I try to update the firmware again on the RTK Surveyor after power off/on the device.

Got the following message. Is it another issue?
```
State: Rover - No Fix
Batt (95%): Voltage: 4.21V Charging: 16.22%/hr Green

beginLoggingUBX: No date/time available. No file created.

SparkFun RTK Surveyor v1.8-Oct  1 2021
** Bluetooth broadcasting as: Surveyor Rover-0CE2 **
Menu: Main Menu
1) Configure GNSS Receiver
2) Configure GNSS Messages
3) Configure Base
4) Configure Ports
5) Configure Logging
6) Display microSD contents
d) Configure Debug
r) Reset all settings to default
f) Firmware upgrade
x) Exit
[ 12393][D][esp32-hal-i2c.c:1371] i2cProcQueue():  Gross Timeout Dead start=0x3064, end=0x3068, =4, max=53 error=1
[ 12393][E][esp32-hal-i2c.c:336] i2cDumpI2c(): i2c=0x3ffbf03c
[ 12397][I][esp32-hal-i2c.c:337] i2cDumpI2c(): dev=0x60013000 date=0x16042000
[ 12404][I][esp32-hal-i2c.c:339] i2cDumpI2c(): lock=0x3ffbc0a8
[ 12410][I][esp32-hal-i2c.c:341] i2cDumpI2c(): num=0
[ 12414][I][esp32-hal-i2c.c:342] i2cDumpI2c(): mode=1
[ 12419][I][esp32-hal-i2c.c:343] i2cDumpI2c(): stage=3
[ 12424][I][esp32-hal-i2c.c:344] i2cDumpI2c(): error=1
[ 12429][I][esp32-hal-i2c.c:345] i2cDumpI2c(): event=0x3ffbc240 bits=10
[ 12435][I][esp32-hal-i2c.c:346] i2cDumpI2c(): intr_handle=0x3ffbc274
[ 12441][I][esp32-hal-i2c.c:347] i2cDumpI2c(): dq=0x3ffec6c0
[ 12447][I][esp32-hal-i2c.c:348] i2cDumpI2c(): queueCount=2
[ 12452][I][esp32-hal-i2c.c:349] i2cDumpI2c(): queuePos=1
[ 12457][I][esp32-hal-i2c.c:350] i2cDumpI2c(): errorByteCnt=125
[ 12463][I][esp32-hal-i2c.c:351] i2cDumpI2c(): errorQueue=1
[ 12468][I][esp32-hal-i2c.c:352] i2cDumpI2c(): debugFlags=0x00000000
[ 12474][I][esp32-hal-i2c.c:329] i2cDumpDqData(): Debug Buffer not Enabled
[ 12481][I][esp32-hal-i2c.c:372] i2cDumpInts(): Debug Buffer not Enabled
No user input received.
```

I retry later  with success (done something else during 10 mn)  
The difference: may be the device updated the time from the F9P ? 🤔

nseidle, 2022-02-12:
FYI - loading firmware files over WiFi may have to go away. Because v2 of the ESP32 core [broke dedicated SPI](https://github.com/greiman/SdFat/issues/349) on sdFat the write time to the SD card is severely reduced. It's still very good for logging but transferring the firmware binary file over WiFi then recording it onto the SD card can take up to 2 minutes. There is likely a way to load new firmware direct over the air (OTA) to the ESP32 but I haven't dug into it. For now, please load firmware via microSD and use the text menu 'Firmware' method.

closed 2022-02-12 by nseidle

nseidle, 2022-02-16:
Starting in v1.10 of the firmware we are doing away with loading to the SD card and loading the firmware directly to the ESP32 over WiFi. This is faster, more reliable, and is a better user experience (IMO). If you feel like giving v1.10 a try, please let me know if you see any issues.