sparkfun/SparkFun_RTK_Firmware #71: Feature request : include NAV messages in UBX preset

Eric-FR, 2021-11-19:
Right now, there are two presets in "Configure GNSS Messages":
7) Reset to Surveying Defaults (NMEAx5)
8) Reset to PPP Logging Defaults (NMEAx5 + RXMx2)

Some NAV messages are very useful. 
- UBX-NAV-SAT: give ephemerides of satellites. Although one may found it somewhere else on internet, being available in the UBX file allows to have a first look at data without external connexion. For post-processing with RTKLib, we don't need to retry an additional file, especially if ephemerides are not included in RINEX from base station.
- UBX-NAV-PVT: the all in one navigation message... "Navigation position velocity time solution".
- UBX-NAV-HPPOSLLH: give higher accuracy on position than PVT (mostly to follow noise in static conditions...)

I don't know what is the best way to implement it. Include it in "PPP Logging Defaults"? Add a new "Extended PPP Logging Defaults" with (NMEAx5 + RXMx2 + NAVx3)? Allow users to define presets?

nseidle, 2021-12-13:
Do you need NMEA as well in this scenario? If it was just RXM+NAV perhaps we could call it 'Reset to UBX Defaults' or something.

I am reluctant to add another button. I have not heard of any other users who need to quickly change between message setups. It's very often 'set it and forget it'. But please convince me otherwise!

Eric-FR, 2021-12-23:
Although, there is some redundancy between NMEA and NAV, having booth provides more versatility i.e. each software may find what he want. The overhead of NMEA is just 20 %.

For my DSLR, I'm using 3 of the 4 available user profiles. Something like Standard, Panoramic capture and Astronomy. I'm using Standard 95% of the time but it's very useful to have the other ones when needed. 
So, for a RTK receiver, I can imagine Car, Bicycle, Pedestrian and Base. But I agree not to add a button. It should be an option in the wifi menu.

nseidle, 2022-02-12:
> For my DSLR, I'm using 3 of the 4 available user profiles. Something like Standard, Panoramic capture and Astronomy. I'm using Standard 95% of the time but it's very useful to have the other ones when needed.

Makes sense. Ok, how should we do this?

>  I can imagine Car, Bicycle, Pedestrian and Base. But I agree not to add a button. It should be an option in the wifi menu.

So *Car, Bicycle, Pedestrian, Base* are different modes the user would choose to enter into? And each mode would have configurable messages that are enabled? Would the modes be accessible from the main display? Hmm. I think that makes sense to me. I can foresee being able to rename 'Rover' to whatever the user choses, and adding up to two additional modes (for you it would be 'Bike' and 'Walk'). This is a lot of code, but do-able. Configure from the text menus is straight forward but  whatever we do, I worry about the AP (access point) config page... Setting up the same functionality in HTML and JS gets really complicated (for me at least, a EE).

Eric-FR, 2022-03-26:
During my last vacations, I used the Express with my car and I also did static tests on surrounding base stations. So, I had to switch dynamic between _Automotive_ and _Stationary_ and check that setting were correct each time I started again. So, I switched to _Portable_ and didn't changed it until the end of the travel.

One way would be that the user stores several profiles, on the µSD, maybe in a dedicated subfolder, and, then, with the Express menu, a sub-menu Presets for choosing one profile. And, it would be nice to display the name of preset at startup.

nseidle, 2022-03-28:
I agree. I have made decent progress on [this branch](https://github.com/sparkfun/SparkFun_RTK_Firmware/tree/UserProfiles1). It's working in the flash memory of the ESP32. I still need to get the file system to 'see' the new file types.

nseidle, 2022-03-30:
I have the basics working:
[image]

Next, I need to get profiles into the display menu.

nseidle, 2022-03-30:
This was a long and arduous journey but I think we now have profiles.

A 'Profile' is a unique copy of **all** the device's settings. Base, logging, receiver, NTRIP credentials, all of it. This should be a handy method for switching between NTRIP casters, among other applications.

There is a maximum of 4 profiles (currently). You can edit the name of any given profile via the 'p' Configure Profiles sub menu. You can select a given profile by pressing the setup button. Once a profile is selected the unit will reset to apply the new settings.

If the user has only one profile created (the default state) then no additional menus are presented.

You can erase all settings ('Erase LittleFS') from within the debug menu (now under the System Status menu). 

I do *not* plan to implement Profile configuration via WiFi AP. It's not hard, just more work. If you (or someone else) requests it, I'll implement it.

[image]

[image]

[image]

nseidle, 2022-03-30:
Binary pushed to [v1.11 release](https://github.com/sparkfun/SparkFun_RTK_Firmware/releases/tag/v1.11) page.

Eric-FR, 2022-04-02:
With my RTK Express, I created 4 profiles. Indeed, when I press the setup button, they don't appear in the menu list. I still see Rover and Base, not my own profiles that are having different names.

RTK_Surveyor_Firmware_v1_11-RCApr1.bin

nseidle, 2022-04-04:
Sorry! My mistake. I implemented for Facet but failed for Express. The latest RCApr4 should have this corrected.

Note: You will still see Rover and Base. Profiles are shown below the Config menu line. I did this because I envision users wanting to modify a profile's state from the button menu. If you think we should just make users select a different profile to select rover or base, I think that's a good idea too.

Eric-FR, 2022-04-07:
Where can I find the RCApr4?

nseidle, 2022-04-07:
The [release page](https://github.com/sparkfun/SparkFun_RTK_Firmware/releases) always has the latest. Looks like I've already posted RCApr5.

nseidle, 2022-04-07:
Under assets:
[image]

Eric-FR, 2022-04-08:
So after update to the last RC (Apr 5), I can see and use my user defined profiles on the menu of the RTK Express.

About the use of wifi to define profiles, I'm not interested in. My logic is to set my profiles at home, when planning travel for instance, and I feel more comfortable using a (desktop) PC rather than a smartphone.

Concerning the removing (or not) of Base and Rover options from the menu when user defined profiles are present, the point is that I don't like interfaces that are changing due to indirect actions. It's a way to lost the user. Imagine a user is creating a new rover profile and don't find the Base option any more.

nseidle, 2022-04-08:
>  I don't like interfaces that are changing due to indirect actions

Ah, I very much agree.

> Imagine a user is creating a new rover profile and don't find the Base option any more.

Well said. I will leave the profiles as is for now. If you can recommend any improvements to the UI, please do so.

nseidle, 2022-04-12:
Feature added and rolled into v1.11 release. Please open new issue if you find anything else.

closed 2022-04-12 by nseidle