sparkfun/SparkFun_RTK_Firmware #72: Feature request : external triggers

Eric-FR, 2021-11-27:
This is a prospective request, maybe for v2.0 by using the qwiic connector.

I want:
- record external event i.e. if an external contact is closed, I want the precise timestamp of that event (in somehow, similar to issue #69).
- generate an external event to trigger an external device. Typically close a contact each second with the 1 pps from F9P to trigger a camera.

nseidle, 2021-11-29:
A fine idea. To be clear, the Surveyor would be difficult, but the Express has a digital mux that makes this very easy via the Data port. Qwiic connector is still possible but a bit more complex.

Benefit of using Qwiic: It would work on either Surveyor or Express. We could potentially auto-detect when the 'trigger' board is present and record events as needed. Con of Qwiic: I would have to do a lot more hardware work. It's not bad, but just another thing to work on.

Benefit of using the Data port: The user could hook up anything using the cable provided with the unit. This would allow the unit to record a TM2 message when an external contact is closed, and toggle an output pin to trigger an external device.

How do you want the user interface to work? I foresee adding to the 'Port' menu. Something like 'Event Recording/Triggering' menu would appear when the ADC/DAC port is selected. Event recording is easy enough - that's a true/false. What about Triggering? The user would need to input when to trigger and under what circumstances. I want to keep it as simple as possible. Would you prefer to enter a 'time between triggers' in milliseconds?

Note: The TM2 message can also be generated when the INT pin is toggled. This can generate **very** accurate time stamps but if we are trigger via a GPIO on the ESP32, the time delay is greater - perhaps 100ms. Do you can if your events are recorded ~100ms after the event? We can certainly build in support for recording events via the INT pin, but, because the mux port has to be either PPS/INT(direct event trigger but no event output) *or* ADC/DAC(event input and event output), the user would need to chose.

If you can describe what you are doing, that would help me understand it. Are you just wanting to do time lapse photos?

Eric-FR, 2021-12-02:
One background is [Mapillary](https://www.mapillary.com/app/). The idea is to take geo-localized pictures while moving along ways. RTK give us a great step forward in term of accuracy. One can see that now,  tracks of pictures are stuck to roads: [sample](https://www.mapillary.com/app/?lat=45.241099902098654&lng=5.85103991283836&z=15.797218473235667&pKey=243422590915546&focus=map). My final goal is to precisely locate by triangulation items visible on several pictures. 
Indeed, when looking carefully in curves or round-about, one can see that longitudinal positioning along tracks is not accurate. I made first investigations with Mapillary android application. The time shift along the track is not stable. I also tried with the external app, Open Camera. Again, not time stable. Overall, it seems that Android is not time stable (not a RT OS).

So, my idea is to use an external device i.e. a digital camera. One can trigger picture recording with the appropriate plug/cable when closing contact on it (like a basic remote control for time-lapse). On can also detect when a picture is taken by checking contact at flash grip (picture bellow).

[image]

When the curtain is opened, the contact on grip is closed and, when the curtain is closed, the contact is opened.

One way would be to use the 1 pps to control a reproducible time-lapse although one will still have to adjust manually some time delay.
The other way would be to use a time-lapse remote and connect the flash grip to F9P INT pin. (I made such an attempt with F9P+adalogger. It was running at home but not in the car, due to my bunch of wires going everywhere...).

Overall, I would say that the second way would be the best. It may also be used for precise astronomic event timing like [asteroid occultation](http://lucy.swri.edu/2021/03/17/Occultations.html) (but F9P is oversized for such a job). Now, concerning the accuracy, if one is driving at 30 m/s (108 km/h or 65 mph), I would like to target 10 cm accuracy. This would give 3 ms. Well, 10 ms (30 cm) at such speed would be enough. You don't think that one can reach such an accuracy when going trough the ESP32? May be, this is not the right way to proceed. My goal is to have a (quite) precise timestamp, not necessary a TIM2 message. Actually, with the Express, when taking a mark, the corresponding NMEA sentence is not self-consistent i.e. we need to analyse other sentences to get the position/time of the mark. Can't we have a precise time in the ESP32? With something similar to [Raspbery NTP server](https://www.satsignal.eu/ntp/Raspberry-Pi-NTP.html)? The PPS signal from GPS is routed to an interruption of the rasp. 

BTW, I didn't noticed that there was a data port so let's go for it.

nseidle, 2021-12-13:
I've added prelim control of outgoing triggers (via the pulse pin) and incoming triggers (via the INT pin). There is a sub menu that can be accessed from the ports menu when PPS/INT is selected for the Data port. Note: Currently I've only implemented serial menu configuration. Once we get the interface nailed down, I'll push to the WiFi AP page as well.

The Pulse pin can be configured to pulse every 1us all the way up to ~4200s (71.5 minutes) between pulses. The pulse duration can also be set (default is 100ms). The pulse polarity can also be set (default is rising, or Low-High for pulse duration-Low). You should be able to use this pin (the green wire) on the data port to drive your camera or anything else. These pulses are *very* accurate (better than 60ns).

The logging of TM2 or incoming interrupts is supported. These are accurate with 10s of ns. Set this option to true and a custom NMEA message is recorded to the log with the Time of week (ms and ns) fractions when the event occurred. For example: 

[image]

    $GNTXT,01,01,01,60,441965,520690*55

This event is the 60th event that occurred at 441.965ms and 520.690ns after the start of the week, or combined 441.965520690ms after the start of the week. Naturally these numbers can get quite long so u-blox splits them up. But I can combine into one print if needed.

I plan to add the TM2 accuracy estimate to the report as well.

I suspect you will want to record a full UBX-TIM-TM2 sentence to the log file so that you can post process. Let me know if this is the case. I will try to support it.

RC attached here: [RTK_Surveyor_Firmware_v110-RCDec-13.zip](https://github.com/sparkfun/SparkFun_RTK_Firmware/files/7707467/RTK_Surveyor_Firmware_v110-RCDec-13.zip)

Eric-FR, 2021-12-14:
I will update firmware and do testing during Christmas vacations. Although, I'm not 100% sure that I will take advantage of ns accuracy.

And, yes, I want the full recording of UBX-TIM-TM2 if UBX logging is activated.

nseidle, 2021-12-15:
Logging the UBX-TIM-TM2 message via an external switch closure on the Data port pin is very doable. I'll work on that.

Please enjoy your vacation time! Let me know how testing goes.

nseidle, 2022-02-12:
Hi @Eric-FR - Please let me know if you've had a chance to test external triggers. I'd like to get it into the v1.10 release.

Eric-FR, 2022-02-17:
Hi Nathan, although it is high in my TODO list, I'm not sure that I will be able to test it soon, i.e. not before March.

nseidle, 2022-02-18:
This is added in [v1.10](https://github.com/sparkfun/SparkFun_RTK_Firmware/releases/tag/v1.10) (full release). I'm especially happy with the WiFi update feature via AP config page. Please give it a try. Closing since I think this is complete. Please open a new issue if you come across any problems.

closed 2022-02-18 by nseidle