sparkfun/SparkFun_RTK_Firmware #74: Feature request : Add firmware information at beginning of NMEA file

Eric-FR, 2021-12-24:
Recently, I faced a bug ([here](https://github.com/sparkfun/SparkFun_u-blox_GNSS_Arduino_Library/issues/94)) and was in trouble finding the origin because I changed a lot of parameters in the same period, including firmware upgrade of the Express.

So, for debugging (and quality control), it would be nice at the beginning of the log file to have one or more NMEA GNTXT sentence(s) showing firmware version, device name, device serial number, F9P firmware version...

nseidle, 2021-12-24:
Agreed. I will add shortly.

PaulZC, 2021-12-24:
Please break the information up into multiple GNTXT messages. If the total length of a single message exceeds 128 characters, it will break the Integrity Checker ([again!](https://github.com/sparkfun/SparkFun_u-blox_GNSS_Arduino_Library/issues/94)).

Merry Christmas!

nseidle, 2021-12-30:
Yup. The text content is [limited to 62 characters](https://www.nmea.org/Assets/20160520%20txt%20amendment.pdf). Total sentence length is 82 characters max.

nseidle, 2022-02-11:
I've got this implemented in v1.10. It currently looks like this:  

    $GNTXT,01,01,00,ESP_RST_POWERON*1C
    $GNTXT,01,01,03,v1.10-Feb 11 2022*54
    $GNTXT,01,01,04,HPG 1.30*36

Note the NMEA types:

    //Custom NMEA sentence types output to the log file
    typedef enum
    {
      CUSTOM_NMEA_TYPE_RESET_REASON = 0,
      CUSTOM_NMEA_TYPE_WAYPOINT,
      CUSTOM_NMEA_TYPE_EVENT,
      CUSTOM_NMEA_TYPE_SYSTEM_VERSION,
      CUSTOM_NMEA_TYPE_ZED_VERSION,
    } customNmeaType_e;

If you'd like to see something else, please let me know.

closed 2022-02-11 by nseidle

Eric-FR, 2022-03-27:
Checked wit a recent file:
```
$GNTXT,01,01,00,ESP_RST_POWERON*1C
$GNTXT,01,01,03,v1.11-Mar 14 2022*4F
$GNTXT,01,01,04,HPG 1.30*36

```

Everything is right (Integrity Checker, UBX Converter, RTKLib).