sparkfun/SparkFun_RTK_Firmware #75: Configuration as Base

rrodas64, 2022-02-03:
Hello:
I have two issues encountered when trying to configure RTK Facet as a fixed base.

when I choose to enter the geodetic coordinates and save the configuration, the rtk facet uses the ECEF data.

Doing tests I have found that, for rtk facet to use the geodetic data, I had to enter the geodetic data first, save it, and then specify that it would be ECEF coordinates and save again. (As it appears in the image).
With this configuration RTK facet takes geodetic coordinates instead of ECEF.
I think it's a programming problem.

Another issue is that when I refresh the rtk facet page in my browser, the device turns off, and I have to try several times to turn it on.
[image]

nseidle, 2022-02-03:
Hi - thanks for reporting!

> when I choose to enter the geodetic coordinates and save the configuration, the rtk facet uses the ECEF data.

Interesting. I'll try to replicate.

> Another issue is that when I refresh the rtk facet page in my browser, the device turns off, and I have to try several times to turn it on.

This is probably related to this [known issue](https://github.com/sparkfun/SparkFun_RTK_Firmware/issues/65). We currently do not support multiple opening of the WiFi config page. It's best to power cycle the unit between configs. Also, if your unit gets into an awkward state press and hold the power button for ~10s, then release. This will power off the unit completely. Then press and hold the power button to turn the unit back on.

nseidle, 2022-02-03:
Ok - I believe I've located a few issues but haven't quite fixed the mall. The Facet was sending the inverted bool for the ECEF setting. Additionally, strtok() was not dealing correctly with two commas in the CSV string coming back from the browser. Also, I've changed the 'Exit to Rover' to 'Exit and Reset' to avoid the problem with the config page begin not reloadable.

The last problem is the unit won't go back to Survey-In. I may need a few more days as I've run out of time. My apologies.

rrodas64, 2022-02-04:
thank you for your immediate response. I'll be waiting for your notice

nseidle, 2022-02-12:
Ok, all is fixed. Please load [v1.10](https://github.com/sparkfun/SparkFun_RTK_Firmware/releases/tag/v1.10RC) onto your unit and let me if you still have problems.

nseidle, 2022-02-18:
This is fixed in [v1.10](https://github.com/sparkfun/SparkFun_RTK_Firmware/releases/tag/v1.10) (full release). Please open a new issue if you come across any other problems.

closed 2022-02-18 by nseidle