sparkfun/SparkFun_RTK_Firmware #76: [sparkfun]RTK Express support

jblee94, 2022-02-03:


jblee94, 2022-02-07:
If then, 
Here's the first question.
In "3) set required means 3D standard deviation: 5.000 meters", what does the 1~5 meters setting mean?

I'm curious about the change when setting 1m and the change when setting 5m.(How to check)

Here's the second question.
If it is set to "base", does the HPA have to be within 1m for the survey to start?
Is there any way to set the search to be done within 2m?

nseidle, 2022-02-08:
> "3) set required means 3D standard deviation: 5.000 meters", what does the 1~5 meters setting mean?

This is the standard deviation that the device needs to be below before it will exit survey-in mode and begin outputting RTCM data. I have run a few tests lowering the standard deviation down to 1m from the default of 5m. The accuracy was improved negligibly. I recommend you stick with u-blox's recommended 5m and 60s.

> does the HPA have to be within 1m for the survey to start?

Yes.

> Is there any way to set the search to be done within 2m?

Not currently. Why do you need to increase this? If you are surveying in an environment where the HPA is > 1m, you will have many inaccuracies after the survey in complete. It is better to wait the few seconds for the HPA to drop below 1m.

nseidle, 2022-02-18:
Closing due to inactivity. Please let us know if you any any other questions.

closed 2022-02-18 by nseidle