sparkfun/SparkFun_RTK_Firmware #77: RTK Facet wifi connection

skleins000, 2022-02-05:
I am having a problem connecting my RTK Facet device to an open wifi source. I have successfully added it to multiple wifi sources that require a password but am unable to connect it to the open source one that has no password.

I am wondering if anyone can help me with this problem

The other question is who I can get the exact coordinates for the fixed base so that I can transmit the precise base station location, in message type 1005 or 1006.  Currently I am only sending: 
Messages 1074(1), 1084(1), 1094(1), 1124(1), 1230(10)

nseidle, 2022-02-08:
Hi - Thanks for reporting. In most of our testing we've been using WiFi APs that had a PW. I'm not too surprised there may be an issue with no password APs. I'll see if I can replicate and fix shortly.

nseidle, 2022-02-08:
> who I can get the exact coordinates for the fixed base

I am confused by this statement. You can obtain the coordinates for your fixed base following the [How to build your own DIY GNSS reference station](https://learn.sparkfun.com/tutorials/how-to-build-a-diy-gnss-reference-station/all#gather-raw-gnss-data) tutorial. Please describe your setup in more detail and I'll try to help. When an RTK product enters Base mode, it automatically enables-for-output the 6 RTCM sentences needed for RTK ([enableRTCMSentences()](https://github.com/sparkfun/SparkFun_RTK_Firmware/blob/main/Firmware/RTK_Surveyor/System.ino#L280)). This includes 1005, 1074, 1084, 1094, 1124, and 1230 (1 per 10s).

skleins000, 2022-02-10:
I am trying to set up a permanent base using the RTK Facet. Right now i have it on a tripod in my room so that i can test everything  before i mount it into the final location. I guess because i havnt used the PPE precise location setup, because i got confused of how i could download the correction data. I you could walk me through the steps that would be great.

nseidle, 2022-02-11:
Do you have an SD card in the unit? Currently the settings file is incorrectly parsing the wifiPW variable when it is zero length, but this only happens with an SD card is used.

skleins000, 2022-02-11:
There is an SD card port but I haven't put an SD card in there yet.

Sent from Nine<http://www.9folders.com/>

nseidle, 2022-02-12:
Ok, blank passwords should be working. Please load [v1.10](https://github.com/sparkfun/SparkFun_RTK_Firmware/releases/tag/v1.10RC) onto your unit and let me if you still have problems.

> i have it on a tripod in my room so that i can test everything before i mount it into the final location

This is both a good idea and a bad idea. Making sure you understand the hardware before going into the field is a great idea. However, GNSS reception indoors is **very** bad. It may not be possible to get a fix indoors. Please experiment with the hardware then move outside so that you can get better reception.

To obtain the precise location of your unit please give the [How to build your own DIY GNSS reference station](https://learn.sparkfun.com/tutorials/how-to-build-a-diy-gnss-reference-station/all#gather-raw-gnss-data) tutorial a read. It will walk you through the steps.

skleins000, 2022-02-15:
I tried doing that but whenever i put the WIFI SSID in and leave the PW blank, whenever i come back in it has automatically added a password. Wondering what i should do about that?

nseidle, 2022-02-15:
Please take and post screenshots of what you're seeing. 

It would help me most if you captured the text menus via a terminal. I cannot replicate your issue using v1.10 firmware. 

Opening of menu showing v1.10:
[image]

Entering a password, then removing it shown below:
[image]

skleins000, 2022-02-16:
How do you access a terminal when on Google chrome?

nseidle, 2022-02-16:
We have a good tutorial [here](https://learn.sparkfun.com/tutorials/terminal-basics) but that's beside the point. Sounds like you are using the browser. I tested via AP config page and replicated your issue. Should now be fixed. Please try [RC Feb-16](https://github.com/sparkfun/SparkFun_RTK_Firmware/releases/tag/v1.10RC).

skleins000, 2022-02-17:
Thanks, that works now. And i will follow the tutorial you sent for get precise positioning.

nseidle, 2022-02-17:
Excellent. Thanks for testing and reporting back. Let us know if you have any more issues.

closed 2022-02-17 by nseidle

nseidle, 2022-02-17:
The Facet is designed for mobile use and is not recommended for permanent mounting. You might be able to seal it with glue but the connector aperatures are not waterproof and the display will not like long term UV exposure. Please use the [L1/L2 antenna](https://www.sparkfun.com/products/17751) (which we've had outside constantly for nearly 2 years in Colorado UV) with a cable running to the [RTK Express](https://www.sparkfun.com/products/18442), or for a cheaper solution, the [ESP32+ZED breakout](https://learn.sparkfun.com/tutorials/how-to-build-a-diy-gnss-reference-station/all#esp32-setup-option-2) is a great way to go.

skleins000, 2022-02-20:
Ok, if i get the RTK Express will it have the same problem as the Facet with connecting with the open Wi-Fi I have here?

nseidle, 2022-02-21:
We just released the v1.10 firmware on Friday Feb 18th. It will take us a few weeks to propagate that firmware to all of our distributors so I cannot guarantee what firmware will be loaded on a unit you purchase today, but you will be able to update the firmware on RTK Express you receive via SD card to v1.10 and fix the problem connecting to APs without passwords.