sparkfun/SparkFun_RTK_Firmware #81: High frequency messages are batched via Bluetooth

Guadelup, 2022-03-01:
### Subject of the issue
High frequency (>10 Hz) messages are batched when transferred via Bluetooth.

### Your workbench
- Sparkfun RTK Express+ with firmware 1.10
- Connected to an android device via Bluetooth using the spp
- Powered via battery

### Steps to reproduce
Enable some kind of high frequency message (e.g. set MSGOUT for ESF-RAW to 1)

### Expected behavior
The data should arrive in near real time (every xx ms).

### Actual behavior
The data arrives roughly once per second in big batches containing the data of the last second.

nseidle, 2022-03-01:
Thanks for reporting. What app are you using to view the data? What phone are you using?

Do you know the approximate size of the message you are trying to send in bytes?

The current firmware supports 4Hz, anything over that is untested. I do not (yet) know the size of the ESF-RAW message size but if it's large, it may be clogging the SPP connection.

Guadelup, 2022-03-02:
The ESF-RAW messages have a size of roughly 90 bytes and are sent with a frequency of up to 100 Hz. Bluetooth should be able to easily transfer this amount of data without major delay.

I've tried multiple Android devices with a custom application to receive the data.
To test possible Bluetooth throughput and possible errors in my receiver app, I've also created an application which rapidly (100Hz) sends packets of size 100B via spp. This does not result in the observed behaviour, but rather in the expected behaviour of receiving data separately (if flushing correctly) or in small batches, which leads me to believe that the problem is in the ESP32 code for batching/sending data via Bluetooth.

nseidle, 2022-03-04:
tldr; Don't use the BT timestamp, use the ESF-RAW message timestamp instead (see below).

[image]

Connecting u-center over Bluetooth to inspect the ESR-RAW packets, I am seeing decent delivery with a message delivered every few ms. Below, I am getting ESF-RAW at 4Hz and all other messages at 1Hz, with a fix rate of 4Hz:

[image]
Above, there is very little batching or delay.

Wow, connecting over USB, these raw readings are coming fast and hard (below)

[image]

I assumed this message was like other messages and would be limited to a report per fix (or less). It's *far* above that.

From the ZED-F9R integration manual: 

> To enable this feature the UBX-ESF-RAW message must be enabled using CFG-MSGOUTUBX_ESF_RAW keys. If any non-zero rate is selected the message will be output at the rate at which the sensors are sampled.

So this means a few things:

* I still don't know the rate of this message but from the USB packet console it looks like 2-3ms per message. That's ~500Hz. At 68 bytes per message that's ~34,000 bytes per second. I'm surprised the ESP32 SPP is working at all!
* It's all or nothing, there is no throttling (I confirmed this).

Looking again at the BT SPP feed I do see some batching:
[image]

Above we see batches of packets arriving at 2.772, 3.005, 3.425, and 3.785. ~300ms gaps between batches. Later, when the link gets statured I start to see what you report:

[image]

With batches having ~900ms gaps.

I'm not sure what else we can do here. The computer or cell phone is going to implement SPP buffering and timing on their end that I don't control. That said, every ESF_RAW packet has a timestamp. I recommend not using the BT timestamp, but the message timestamp instead.

If you've got any good ideas, I'm all ears. ESF-RAW is working and being transmitted well. The report rate and the nature of SPP is going to prevent any predictable delivery timing.

Guadelup, 2022-03-10:
After some further investigation I actually think the batching does not happen in the spp part.

`serialGNSS.setTimeout(50); `

Seems to have the effect that if during these 50ms new data arrives, the timeout begins anew. With rate higher than 20 Hz it therefore doesn't stop reading until the buffer is full (Which for ESF-RAW seems to be after ~900ms).

I don't know if the timeout is necessary for other parts of the software but a quick test with a custom app (with the timeout set to 0) which does nothing more than send data between SerialBT and SerialGnss indicates that the described problem no longer occurs.

nseidle, 2022-03-10:
Hmm, very good point. And good job digging through the firmware.

Below is the ESF RAW over bluetooth with the original 50ms timeout (we see the expected batching):
[image]

Below is with the timeout set to 0ms:
[image]

I stand corrected. It looks very improved!

The original setTimeout of 50ms was set a long time ago and I believe it was selected to reduce ESP32 guru panics (system resets) and logging corruption. Most of these issues have been worked out and *should* be unrelated to the SPP serial transfer. Attached is a v1.11 dev firmware if you'd like to try it out (timeout set to 0).

[RTK_Surveyor_Firmware_v1_11-RC-Mar10.zip](https://github.com/sparkfun/SparkFun_RTK_Firmware/files/8225650/RTK_Surveyor_Firmware_v1_11-RC-Mar10.zip)

I will do some additional testing to make sure the rest of the system is still stable.

nseidle, 2022-03-14:
Excellent. Thanks for your help and testing it out. This feature will make it into production in the coming weeks.

closed 2022-03-14 by nseidle