sparkfun/SparkFun_RTK_Firmware #84: RTK Express powers off before displaying splash screen

LeeLeahy2, 2022-03-17:
### Subject of the issue
Pushing the power button on the RTK Express does not always power on the device.  I believe that this is due to button bounce.  I am verifying that the ESP32 starts running by connecting to the CONFIG ESP32 port.

The v1.10 code requires that the user hold the Power button on without any bounce for just over 500 mSec.  This also does not take into account the processing time associated with bringing up the GPS, calls to beginGNSS and beginEEPROM as well as determining the product.  If any bounce occurs in this 500 mSec then the RTK Express will shut off.

I have changed the powerOnCheck routine to the ignore the bounce during the 500 mSec and check for the power button being depressed after the delay.  See the attached file.

### Your workbench
* What version of RTK Surveyor firmware are you running? v1.10
* How is everything being powered? Battery, USB serial

### Steps to reproduce
Push the power button, not real hard
[powerOnCheck.txt](https://github.com/sparkfun/SparkFun_RTK_Firmware/files/8289582/powerOnCheck.txt)

### Expected behavior
The device should power on and display the splash screen

### Actual behavior
The device powers on and then turns off before displaying the splash screen

LeeLeahy2, 2022-03-30:
//User has pressed the power button to turn on the system
//Was it an accidental bump or do they really want to turn on?
//Let's make sure they continue to press for 500ms
void powerOnCheck()
{
  powerPressedStartTime = millis();
  if (digitalRead(pin_powerSenseAndControl) == LOW)
    delay(500);

#ifndef ENABLE_DEVELOPER
  if (digitalRead(pin_powerSenseAndControl) != LOW)
    powerDown(false); //Power button tap. Returning to off state.
#endif

  powerPressedStartTime = 0; //Reset var to return to normal 'on' state
}

nseidle, 2022-04-01:
Hi Lee - I cannot replicate your issue. Is this happening to you often? Have you tested your solution and seen improvement?

I'm very hesitant to change the powerOnCheck as it may have unintended consequences for a lot of users. Your (2nd) recommendation is ok, but you're not really debouncing as much as you are checking if the button was pressed a 2nd time sometime before 500ms has expired.

nseidle, 2022-05-17:
Hi Lee - is this still an issue? I haven't seen the need to change powerOnCheck.

nseidle, 2022-05-19:
A fine change. Added to RC.

closed 2022-05-19 by nseidle