sparkfun/SparkFun_RTK_Firmware #85: Base correction from a Sparkfun_RTK_Express or SparkFun_RTK_Surveying to Sparkfun_RTK_Express_Plus

cvjorodn, 2022-03-18:
### Subject of the issue
I have some Sparkfun_RTK_Express_Plus and I would like to improve the accuracy by using a base unit. However, because these units have a GNSS chipset (ZED-F9R), it does not support the base configuration. I would like to know if is possible to provide data correction to the RTK Express Plus with the use of another unit like the Sparkfun RTK Express (https://www.sparkfun.com/products/18442?_ga=2.172766371.1671291917.1647294910-1224788823.1642199496) or the SparkFun RTK Surveying Kit (https://www.sparkfun.com/products/17370)?

nseidle, 2022-03-18:
Hi - You are correct. The ZED-F9R can receive corrections and improve its accuracy and achieve RTK (~14mm HPA), but it cannot generate RTCM corrections (it cannot enter base mode). 

The RTK Express, RTK Surveyor, and RTK Facet all use the ZED-F9P, which can be put into base mode to generate the RTCM, which can then be provided to the RTK Express Plus.

Let us know if anything about our informational materials was not clear. We never want to confuse or lead a user on. I am happy to add additional info to a page or tutorial as needed.

closed 2022-03-18 by nseidle

cvjorodn, 2022-03-18:
Hi,
Thanks for clarifying this. 
The confusion came with the Hook-up guide being the same for all units, including the photos, and only at the base setup section (bottom of the guide), a sentence differentiates the Express Plus from the Express. Also, in terms of marketing,  a "Plus or +" would usually reference a device with more capabilities unless I'm missing something else.

Is there a difference in terms of benefits and limitations between the RTK Express, RTK Surveyor, and RTK Facet to generate RTCM? 
Also, besides a telemetry radio to transmit the corrections, is there an additional component I should consider?

cvjorodn, 2022-10-11:
Hi Nathan,

Thanks for clarifying this.

The confusion came with the Hook-up guide being the same for all units,
including the photos, and only at the base setup section (bottom of the
guide), a sentence differentiates the Express Plus from the Express. Also,
in terms of marketing,  a "Plus or +" would usually reference a device with
more capabilities unless I'm missing something else.

Regards,

Jorge

*Jorge Rodriguez, ** Ph.D., **MSc, **BSc, E.I.T*
Geotechnical Engineer
***@***.***
[image: View my profile on LinkedIn] View my profile
<http://ca.linkedin.com/pub/jorge-luis-rodriguez-navarro/49/b66/98b>