sparkfun/SparkFun_RTK_Firmware #86: Long Term Log

cvjorodn, 2022-03-30:
### Subject of the issue
I'm trying to test the Sparkfun RTK Express Plus to monitor displacements over long term (>1Year). However, the current firmware only allows up to 2880 min of log. I noticed that a related topic was addressed on a previous issue regarding cyclic logging. is this something that could be added or is there already a setting for this?

### Your workbench
Currently, my RTK Express Plus units are running with the Firmware V1.8, and the ZED-F9R Firmware is HPS 1.00.
Data is being recorded at 1 Hz in a microSD card.

### Steps to reproduce
Attached Setting file.

### Expected behavior
Continuous recording of the position.

### Actual behavior
Stops recording data.

[SFE_Express_Plus_Settings.txt](https://github.com/sparkfun/SparkFun_RTK_Firmware/files/8382819/SFE_Express_Plus_Settings.txt)
d

nseidle, 2022-03-30:
I love this idea and it scares me at the same time. The best!

Short answer: Yes, cyclic logging is currently in the release candidate (you can try it today).

Long answer: The current limit is 48 hours but we can easily increase this to 2 years. I would have warned against this because the likelihood of corruption is higher, but with cyclic logging, it should be mitigated to possibly loosing only the last log file.

nseidle, 2022-04-01:
This has been increased to 2 years in [v1.11 RC](https://github.com/sparkfun/SparkFun_RTK_Firmware/releases/tag/v1.11).

Eric-FR, 2022-04-03:
I will face a similar issue for a measurement campaign in a place far far away. I will setup a reference station for the duration of the campaign i.e. 10 days. I will use a second receiver as rover. Due to lake of cellular network and intervisibility between base and rover, I will not be able to use RTK. Instead, I will use post-processing.

I checked with last RC version and I was able to set max duration to 20.000 minutes (~14 days).

nseidle, 2022-04-04:
Great - thanks @Eric-FR!

@cvjorodn - Please give it a try and let us know if you need anything else.

closed 2022-04-04 by nseidle