sparkfun/SparkFun_RTK_Firmware #89: GPS Locked - No Satellites?

LeeLeahy2, 2022-04-03:
### Subject of the issue
Confusing display on screen, RTK Express display has:
* Flashing satellite dish icon
* Zero satellites in view
* Horizontal accuracy 2.39 meters

System menu displays: SIV: 28, HPA (m): 2.353, Lat: 21.356976459, Lon: -158.081948173, Altitude (m): 229.5

### Your workbench
* What version of RTK Surveyor firmware are you running? RTK_Surveyor_Firmware_v1_11-RCApr1.bin
* Are you connected to the device over Bluetooth? No, connected via USB to Config ESP32 port
* What app? minicomm  (terminal emulator)
* Are you transmitting NTRIP back to the device?  No
* How is everything being powered?  Battery / USB
* Are there any additional details that may help us help you? NoRTK_Surveyor_Firmware_v1_11-RCApr1.bin

### Steps to reproduce
1. Power on the system
2. Let GPS sync up
3. Reload the firmware with RTK_Surveyor_Firmware_v1_11-RCApr1.bin

### Expected behavior
1. Satellite lock icon should stop flashing when one or more satellites are in view
2. Satellites in view should be updated when the number of satellites in view changes

### Actual behavior
1. Satellites in view in system menu > 0
2. Satellites in view on screen is zero
3. Satellite dish icon is flashing

nseidle, 2022-04-04:
Ok - I think this is fixed in RCApr4.

nseidle, 2022-04-12:
Fixed and rolled into v1.11 release. Please open new issue if you find anything else.

closed 2022-04-12 by nseidle