sparkfun/SparkFun_RTK_Firmware #90: Logging disabled - Writing to SD card?

LeeLeahy2, 2022-04-03:
### Subject of the issue
Confusing display, RTK Express display shows writing to SD card, SD card icon with bars filling it up.  Logging menu shows logging is disabled.

### Your workbench
* What version of RTK Surveyor firmware are you running?  RTK_Surveyor_Firmware_v1_11-RCApr1.bin
* Are you connected to the device over Bluetooth? No, connected via USB to Config ESP32 port
* What app? minicomm  (terminal emulator)
* Are you transmitting NTRIP back to the device? No
* How is everything being powered? Battery / USB
* Are there any additional details that may help us help you? No

### Steps to reproduce
1. In Rover mode
2. Turn on logging if it is disabled
3. Exit all menus to allow logging to start
4. Enter main menu
5. Enter logging menu
6. Enable logging
7. Exit logging menu
8. Exit main menu

### Expected behavior
When logging is disabled, the SD card menu should be removed and the increasing bar should be displayed.
When logging is enabled, the increasing bar should be replaced with the SD card with bars filling it up.

### Actual behavior
SD card icon and bars filling it up continue to be displayed after logging is disabled.

nseidle, 2022-04-04:
This is likely related to the new SD quick tests. I'll have a look.

nseidle, 2022-04-04:
Ok - I think this is fixed in RCApr4.

LeeLeahy2, 2022-04-05:
The issue still occurs in RCApr4.  Prior to a system reset, the SD card icon continues to have the bars filling the SD card after logging is disabled.  Once the system is reset, it figures out that logging is disabled and displays the increasing vertical line icon.

nseidle, 2022-04-05:
Ah, I misunderstood the issue. Found the problem. RCApr5 has the fix.

closed 2022-04-06 by LeeLeahy2